arendst/Tasmota #309: Support for Shutter motors?

HansWurst90, 2017-03-30:
First of all, thank you for the great firmware. I flashed my sonoff basic and am very pleased with how everything works.

In the Wiki it says, that the iTead Motor Clockwise/Anticlockwise is supported.
With the original firmware it only supports the commands "Clockwise/Anticlockwise".
Are you able to Stop the Motor with your firmware? I asume the two relays are never "on" at the same time when it changes form Clockwise to Anticlockwise.
I want to use that board to control shutters. They use a motor with limit switches.

It would be awesome if there were commands like "Clockwise, Anticlockwise and Stop" with a "safe" transition between the two directions.

arendst, 2017-03-30:
The iTead motor clockwise/anticlockwise uses a dedicated chip to control the relays. The software has no control over the relays as such. All you can do is send a 0 for relayA On/ relayB Off and a 1 for relayA Off/relayB On.

HansWurst90, 2017-03-30:
Oh, thats a shame.
Are you aware of any boards that offer such a feature?
1 board with 2 realys and 3 settings [(1on, 2off), (1off, 2on), (1off, 2off)]

It seems like such a simple thing to control a shutter, garage door or blinds motor but I can't find something fitting.

arendst, 2017-03-30:
You could use a Sonoff Dual and control the relays using your Home Automation tool

HansWurst90, 2017-03-30:
I thought about that. But there would be no mechanism that makes sure that the two relays are never switched on at the same time.
This situation would short the motor if I'm not mistaken.
I would rather want a low level mechanism that makes sure this situation can't occur than to rely on the Home Automation System to do that.

HansWurst90, 2017-03-30:
One last question and then I will stop bothering you.
I am not very experienced with the Sonoff and esp firmware.
But is it possible to some how link the two relays in the Sonoff Dual?
Like when I activate relay 1 the Sonoff turns off relay 2 first?
Or are the possibilities of firmware flashing only limited to the "communication part" of the Sonoff, not the logic of it.

arendst, 2017-03-30:
It could be done but Sonoff Dual uses the serial interface to control the relays and there might be something going wrong on that interface.

Another option is the Sonoff 4CH or the ElectroDragon but again there is currently no code in my software to switch one relay off before switching another relay on.

Could be made but needs research. Someone might provide a PullRequest.

smadds, 2017-03-30:
Regarding interlock safety:
If you could get a relay with changeover contacts (1 pole, 2 way) you could make the switching foolproof. By wiring one relay (single pole single way as in sonoffs) in series with the common of the 2-way one you could never have both outputs live at the same time. 

[image]

davidelang, 2017-03-30:
#288 was created to cover this and related use cases. Overall it's more complicated than your request, but I think as you dig into what you actually need, you'll find that you need more than you think :-)

It's not just the motor direction and start/stop, you also need some way of knowing when to stop running the motor in each direction (limit switches), and probably want a way to only move part of the way (at least a simulated encoder)

288 wasn't intended to be implemented all at once, but rather to outline a framework so that when someone starts implementing it, they can do so in a way that can readily be expanded to cover the full capabilities.

HansWurst90, 2017-03-30:
It takes of course way more than just relays to build a whole shutter system with limit switches and the possibility to to move them part way.

But in some cases, like in mine, you can just replace 2 simple switches (up/down) in the wall with two relays. The shutter have limit switches built in and stop in the open and closed position and part way closing/opening is not a requirement by me.

I understand though, that the Sonoff hardware is nut built for that use case and that you want to build it "the right way".

I am new to the whole topic and enjoy figuring everything out and to learn the possibilities and limits of the whole DIY Home Automation community.

davidelang, 2017-03-30:
so with your shutters, what happens if someone hits both buttons at the same 
time?

If you have limit switches, then you could get away with a SPDT relay (like is 
on the cw/ccw device) and just hook one button to ech of the poles, with 
whatever they switch to on the common (you may need to create a way of powering 
the sonoff device)

HansWurst90, 2017-03-31:
The switches build in a way they mechanically can't be switched on an the same time.
That's why I am looking for a fool proof 2 relay solution.
The Problem with the cw/cww device is, that it only has these 2 states. It can't stop.
I said I don't need automated part way closing/opening, but it would still be nice to be able to stop the motor.

Like I said. With my little experience in this field I did not expect this to be such a complex problem, but I guess I was wrong.
Thank you for pointing all these things out and making suggestions. I won't stop looking for a solution, but I understand the problem of this Idea a lot better now.

davidelang, 2017-03-31:
a foolproof way of doing two directions and stop is to have one relay that 
controls on/off feeding to a second relay that determines the direction.

There was a picture of a paper diagram posted earlier that shows this.

David Lang

ageorgios, 2017-03-31:
I have created a firmware for ESP8266 with http API targeting the Sonoff Dual, but recently learned that I do not have two GPIO Pins free to use with an external switch.

I have created API with /up /down and protects from both firing simultaneously.

Have not tested yet, waiting for my Sonoff Dual, but need to solve the pins problem to integrate two push buttons too.

smadds, 2017-03-31:
Most blinds and projector screens have limit switches and respond to a short pulse to take them all the way up or down. Using the circuit I sketched, set the direction relay (on the right) to up or down, then pulse the power relay (on left) on for a second. The device should go all the way up or down. You only need to hold the power on if you want to partially open or close the device.

davidelang, 2017-04-23:
closing this issue as it seems this discussion has wound down. please reopen it if you feel it needs to continue.

closed 2017-04-23 by davidelang

gekkie, 2017-05-17:
Sorry to re-open this thread.

I needed to use a SonOff Dual to drive a motor for blinds. To make sure I did not burn out the motor I added the following code:
```
sonoff.ino
1600,1606d1598
<     // Motor block: If attempt is made to switch on relay 0 and 1
<     // then turn both relays off
<     if (power == 0x03) {
<         power = 0x00;
<     }
<
<
```

Basically this switches off all the relays if attempts are made to turn 2 relays on that the same time.

HansWurst90, 2017-05-23:
To me that sounds like the solution I was looking for.

Kurvi, 2017-05-23:
WOW, since few days I am also searching for this.
It seems this is also my solution.

One question, where should the code be implemented, in line 1600?

[image]

gekkie, 2017-05-23:
Sorry, long week. File is sonoff.ino around line 1600

```
if (state <= 2) {
    if ((blink_mask & mask)) {
      blink_mask &= (0xFF ^ mask);  // Clear device mask
      mqtt_publishPowerBlinkState(device);
    }
    switch (state) {
    case 0: { // Off
      power &= (0xFF ^ mask);
      break; }
    case 1: // On
      power |= mask;
      break;
    case 2: // Toggle
      power ^= mask;
    }
    // Motor block: If attempt is made to switch on relay 0 and 1
    // then turn both relays off
    if (power == 0x03) {
        power = 0x00;
    }

    setRelay(power);
```

Kurvi, 2017-05-23:
Thank you so much.
:-)

dordagani, 2017-06-03:
Hi guys,
How to inserts this code to sonoff dual?
I'm new in this world of sonoff :)

Kurvi, 2017-06-07:
Here are some Links.
https://www.youtube.com/watch?v=_389pQPLyr8&t=735s
https://github.com/arendst/Sonoff-Tasmota

roondar, 2017-06-11:
Or just waiting the merge code of Pull Request #519

HeHaHu, 2017-06-24:
I use the Sonoff Motor Clockwise/ Anticlockwise device for my shutter. The motor of the shutter has limit switches and automatically switches of. So, there is no need for a "stop" function. 

The only problem now is that the product is obsolete. I don't know what to buy now.

stefanbode, 2017-08-01:
I do have a similar request to open and close valves and one to open/close my marchioness. I use the pulsetimer and calculate upfront the running time. I send the pulsetime and relay to ON. This ensures the relay go OFF again after working time. For the Marchioness I reset the pulsetime to 0.1sec after successful execution, because I had some issues with retained MQTT commands executed twice.

areid1234, 2018-02-16:
Hi all,

Sorry to start this discussion again!

I have a motor which raises and lowers a TV. The TV turns on when raised, the motor stops at the top and bottom of the lift.

I would like to add a Sonoff/Sonoff Dual to the circuit. I saw on a previous note in this thread that a dual could switch "on" the up and switch "on" the down. I have already bought a Sonoff with a singular relay and have placed this within the circuit in several places but with no luck!

I don't know what the brains of the board do but it effectively uses the RF transmitting to a remote to move this up and down.

I am a complete novice but i anyone has any advice and which wires to connect to the Sonoff to get this going I would be grateful! I have also ordered an RF version which could work but I would prefer this wired in

Thanks!

[image]

[image]

[image]

[image]

gekkie, 2018-02-16:
Sonoff with a single relay will only switch on or off. You need at least the dual relay version.

giig982, 2018-02-16:
Moreover, you have to bypass the controller, or work in parallel, non only cut its signals.

HeHaHu, 2018-02-16:
Could this be of any help:

https://www.itead.cc/wiki/Sonoff_4CH_Pro#Working_Mode

areid1234, 2018-02-16:
I agree, I have ordered a Sonoff dual to turn “on” each movement individually. Really can’t see where I could place it. Could it work by re-wiring the DIN to the Sonoff?

Giig982 - do you think I will always have trouble with the controller being there? Any way I could re-wire from what you can see on my diagram?

giig982, 2018-02-16:
In the schema I cannot understand where you cut the wires to install sonoff: before or after the control unit. If before, you can only turn the control unit on or off. In the latter case, you're cutting signal to the motor, only one of two movements, i suppose. And I hope they're at 220V, otherwhise Sonoff won't even turn on. 
You should provide a more detailed schema.

smadds, 2018-02-16:
I don't think you should cut any wires. If you can emulate a prod on switch using a pair of relays connected to the back of the up/down switch that would work.

Do you have a voltmeter to measure the volts from the centre pin of that switch to one of the other pins? Need to know whether that is pulsing mains or a lower (ac or dc) voltage. 

Alternatives could include emulating the remote control transmission or hard-wiring relays across a (spare) remote controller. These options would also be electrically safer and not invalidate warranty on the main unit.

0x3333, 2018-06-07:
I know this issue is closed, but I'd like to let others know what I did to get a NO/NC relay in Sonoff 4CH(Not the pro version) to control a roller blind and other sensors.

In my case, I wanted to use a 4CH basic version to read a BME280 sensor, PIR sensor, be able to control a roller blind motor and my room lights.

The BME280 was connected to the GND/RX/TX/VCC in the available pin headers(Configured as I2C pins). The PIR was connected to the IO2 in the same pin header. The motor required a little change, I removed the CH4 relay and got the 5V and relay signal from the removed relay pins(I used a 5V relay board from china). The GND I got from the capacitor on the supply line. I isolated the relay board(Cannot be seen on the pictures) and used a double sided tape to "glue" it on position. The terminals from the added relay I externalized using plain cables. The common from the added relay comes from relay 3 output, this way I can "turn off" the motor using relay 3 and relay 4 controls the up/down move.

Hope it helps.

[image]

Relay removed and signals soldered.

[image]

Tested

[image]

Mounted inside the case

[image]
Final result