arendst/Tasmota #310: Wemo/hue emulation question

edmogeor, 2017-03-30:
I want to purchase multiple sonoff basic switches and install this firmware on all of them and control them all with Amazon echo. I read on another thread that i would just need to enable the wemo emulation so that the Alexa could recognise it. Will this work with multiple sonoffs? And do I need to use an MSQTT broker on a Raspberry Pi  for example for this setup to work? I'm assuming I just need to enable wemo emulation and change the names on each sonoff so that they appear differently to the echo.

Thanks #

davidelang, 2017-03-30:
I haven't used amazon or google devices to manage things, but people are 
reporting success with both wemo and hue emulations (hue is needed for multiple 
switches/relays see #296)

they should detect all sonoff devices with the emulation enabled.

edmogeor, 2017-03-30:
Hi David. I saw that thread however i was confused as to whether it meant multiple separate sonoffs or devices with one ESP8226 that controls more than one relay. Is it right that i shouldn't need a  MSQTT broker and it should work independently. Thanks

davidelang, 2017-03-30:
yes, you should be able to use the emulation without using mqtt (make sure you 
disable it so that it doesn't keep trying to connect)

that thread was about multiple switches/relays on a single sonoff, but multiple 
sonoffs should work with either emulation as long as you only have one 
switch/relay on them

edmogeor, 2017-03-30:
How would i disable mqtt? Thanks

davidelang, 2017-03-30:
look for MQTT_USE in user_config.h

edmogeor, 2017-03-30:
And change the value to 0? Or just comment this line out?

davidelang, 2017-03-30:
change it to 0
k

altelch, 2017-03-31:
Hi,

yes, you can use multiple Sonoffs in Wemo Emulation mode. The Echo 
supports up to 16 Wemo devices AFAIK and 16 HUE bridges.

Greetings,

      Heiko.

edmogeor, 2017-04-02:
A few more questions. Does the sonoff work with the hue/wemo app and will it definitely work with mqtt disabled. Thanks

altelch, 2017-04-02:
Hi,

the Wemo-Emulation doesn't work with the Wemo App. The old Hue-App works 
with the Hue-Emulation. You can switch on/off and controll brightness/color.

Greetings,

      Heiko.

stefanbode, 2017-04-05:
I have checked both Wemo and Hue with Alexa and also use MQTT SSL for automation control. As long as you have only ONE Relay Wemo is also fine, but I have found out, that the HUE emulation is much more sophisticated and also capable working with e.g. four relays. You have to define the "friendly name" in then "Configuration other" page. For me, it works OOTB like a charm, and I can recommend using HUE.

edmogeor, 2017-04-05:
Thanks for the info stefanbode, the Hue emulation interests me as I would still be able to use the old hue app for control if i didn't want to use Amazon Alexa. I am hoping it works without using a MQTT as then I will not have to use extra hardware such as a raspberry pi for the broker. I know sonoff already have an Alexa skill but you have to say 'ask ewelink, turn the TV off' which seems like a mouthful to me. Cheers for the help guys, will report if it works without using mqtt once my sonoffs have arrived.

davidelang, 2017-04-23:
closing this issue as it seems the discussion has wound down. please reopen if you think it needs to continue

closed 2017-04-23 by davidelang

uros76, 2017-04-29:
Hi. I am trying out Wemo and Hue emulation to be used without MQTT and with Alexa. 
With old Hue app I can control Hue emulated sonoff but then how to connect that with alexa in case I dont have Hue bridge? Any suggestion or I need a Hue bridge?

altelch, 2017-04-30:
Hi,

no, you need no Hue Bridge. The Sonoff emulates the Bridge. Just search 
devices with your Echo and the sonoff should be found or better the 
emulated devices connected to it.

Greetings,

     Heiko.

uros76, 2017-04-30:
I am trying to do exactly this, no luck :(. Alexa does not find HUE emulated sonoff. 
Strangely I can pair emulated sonoff with 1st gen HUE app on phone. This way it works well but still Alexa doesn't discover it. 
I have Hue skill enabled on Alexa, anything else I need to setup on Alexa?

altelch, 2017-04-30:
Hi,

you don't need the HUE Skill. The Echo built in device support is 
enough. Never installed this skill. Try remove and enable debug level 4 
to see what happens during device search.

Greetings,

     Heiko.

uros76, 2017-04-30:
Ok removed all unnecessary skills. Still doesn't recognize emulated HUE sonoff. 
Sorry for asking never did this, how do I enable debug level 4 on alexa?

altelch, 2017-04-30:
Hi,

on Sonoff, not Echo.

Greetings,

    Heiko.

uros76, 2017-04-30:
Thanks for the tip, it did shed some light whats going on. Some progress made, getting closer. 

Let me explain situation. 
I have ISP modem setup as router and DHCP → Asus router with stock firmware setup as wifi AP → DDWRT powered Wifi AP connected through Asus AP to extend the range. I am trying to connect 4 sonoff switches with HUE emulation. 
When a sonoff switch connects to Asus AP it gets recognized with Alexa. Finaly :). 
If sonoff connects through second AP then it doesn't get recognized by Alexa. On the DDWRT I have all security off and even tried putting DMZ on one switch. Simply doesn't connect. 

Any idea how to mitigate this problem? Are there range of ports that Alexa uses and I could just try to forward them?

altelch, 2017-04-30:
Hi,

I'd guess a Problem with multicast traffic. Maybe it's not forwarded by 
one of your APs.

Greetings,

    Heiko.

uros76, 2017-04-30:
Ok in this case I am pointing finger on Asus stock firmware as it doesn't have any firewall setting once its in AP mode. 
Loading DDWRT on it now and will report back if this fixed the problem.

davidelang, 2017-04-30:
we sould need more information about your network setup, are the two APs on two 
different SSIDs? d you bridge the two networks or route things between them? do 
you have NAT between the two networks?

etc.

uros76, 2017-04-30:
Managed to rework the 2 AP's setup and now all sonoff's work as HUE emulated. Alexa discovers them, happy days :). 
The problem I couldn't solve was having 2nd AP wireless bridged to 1st one. This caused lack of multicast traffic forward as Heiko pointed out. Rather than deep diving into network settings on both AP's I simply installed extra cable between them. 
Thanks a lot all for help.

mestes136, 2018-01-28:
Ok this seems to be the exact problem I’m having too. My set ups a bit different. Linksys 6300 for router cat 5 to dlink in bridge mode for WiFi AP and wireless extender (the one I am connecting the sonoffs through) ST will detect only one and then both fight for control. How exactly did you solve this?

uros76, 2018-01-28:
I couldnt get them working through WiFi extenders or WiFi bridged routers. Some traffic gets lost in such setup unfortunately. Works for me extended through cable.

mestes136, 2018-01-28:
What do you mean cable? Like a cable to a WiFi router that the sonoff connects to or a cable to the sonoff?

uros76, 2018-01-28:
Cable to router or extender.

oywino, 2019-09-18:
Hi All,  I have flashed my Sonoff Basic with the Tasmota 6.6.0 firmware and have it up'n'running on my WLAN. Then I disabled MQTT and enabled WeMo emulation, hoping to be able to see the Sonoff in my Home Assistant 0.98.2
The integration shows up automatically in HA, but there are no devices visible.
So, what am I missing here?