arendst/Tasmota #311: Setting webpassword through MQTT does not work if start not with a number

stefanbode, 2017-03-30:
I can reproduce that setting the webpassword through MQTT seems to be dependent on the fist character.

hm/setting/ESP_D67DDD/webpassword -> '4711' is working
hm/setting/ESP_D67DDD/webpassword -> 'a4711' is not working (Result is: hm/status/ESP_D67DDD/RESULT --> {"WebPassword":""}
hm/setting/ESP_D67DDD/webpassword -> '4711a' is working

It looks like a character at the beginning is the problem. A number works fine.

arendst, 2017-04-01:
I can't reproduce. I used both MQTT-spy and mosquito_pub but in all cases it accepts 'a4711' as a valid password:
```
mosquitto_pub -t cmnd/wemos5/webpassword -m a4711
```
```
11:51:22 CMND: webpassword
11:51:22 RSLT: DataCb Topic wemos5, Group 0, Index 1, Type WEBPASSWORD, Data  ()
11:51:22 MQTT: stat/wemos5/RESULT = {"WebPassword":"a4711"}
```
What tool do you use to send mqtt messages?

stefanbode, 2017-04-04:
Quite the same: mqtt-spy-0.6.0-beta-b51-jar-with-dependencies.jar
Let me double check. I'm using TLS and therefore quite short of memory in general. could be a side effect.

stefanbode, 2017-04-04:
Using DEBUG I can see the message is transmitted correctly, but as you see the password is empty...

16:35:17 RSLT: Receive topic hm/setting/sonoff/webpassword, data size 5, data a1234
16:35:17 RSLT: DataCb Topic sonoff, Group 0, Index 1, Type WEBPASSWORD, Data a1234 (A1234)
16:35:17 MQTT: hm/status/sonoff/RESULT = {"WebPassword":""}
16:35:18 Config: Saved configuration (1348 bytes) to spiffs count 21

arendst, 2017-04-04:
What version are you using? I still can't replicate. Are you able to change the password using serial commands?

Of topic: I see you are using spiffs. Spiffs has been removed since version 4.1.1 as it takes up a lot of space and provides no real advantage; my flash routines use two pages anyway. I will not go in discussions about advantages of spiffs :-)

stefanbode, 2017-04-04:
I can change the password using the serial and able to change it using the webinterface. Both works.
I did more investigations and found following root case:
sonoff.ino (805)
int16_t payload = atoi(dataBuf);     this results in payload =0 (databuf = 'a1234') the rest is clear. Password will never set, if payload=0. I will disable spiff and upgrade to 4.1.1. Let's see if there is a change.

stefanbode, 2017-04-04:
ok. 4.1.2 already addressed the problem. The code has changed here and the payload=0 does not have the impact again. Fixed with the remove of "If payload ==0 ...."

closed 2017-04-10 by arendst