arendst/Tasmota #312: counter for operating time of the power switch

raisude, 2017-03-30:
Is it possible to implement a counter how long the power of the relay has been on? Would be great for statistics to know how long the items connected to the sonoff were operating.

Regards Rainer

citrusfizz, 2017-03-30:
I use Node-red and MQTT to track how long my bathroom fan has been on and if its over 20 minutes, sends a message to shut it off.    I know this isn't what your asking for, but i thought i'd share

davidelang, 2017-03-30:
given that any counter the sonoff keeps will be thrown off if it looses power, 
you are probably better off tracking this in your home automation software.
k

davidelang, 2017-04-23:
closing this issue as it seems the discussion has wound down, please re-open if you think it needs to continue

closed 2017-04-23 by davidelang