arendst/Tasmota #313: I got problem when compile with WS2812 DMA

hamwong, 2017-03-31:
I got problem when compile with WS2812 DMA, It is fine to compile without DMA (just WS2812 and Ctype), I set user_config.h as below, I also tried add 1 after #define USE_WS2812_DMA, result same
any help?

```
#define USE_WS2812                               // WS2812 Led string using library NeoPixelBus (+8k code, +1k mem) - Disable by //
  #define USE_WS2812_CTYPE     1                 // WS2812 Color type (0 - RGB, 1 - GRB)
  #define USE_WS2812_DMA                         // DMA supports only GPIO03 (= Serial TXD) (+1k mem)
```

I got below error

```
Arduino: 1.8.0 (Windows 7), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 1M (64K SPIFFS), ck, Disabled, None"

C:\Users\Ham\AppData\Local\Temp\arduino_modified_sketch_10798\support.ino: In function 'void osw_osWatch()':

support:37: error: void osw_osWatch() causes a section type conflict with static volatile void NeoEsp8266DmaMethodBase<T_SPEED>::i2s_slc_isr() [with T_SPEED = NeoEsp8266DmaSpeed800Kbps]

 void ICACHE_RAM_ATTR osw_osWatch(void)

                      ^

In file included from C:\Temp\arduino-1.8.0\portable\sketchbook\libraries\NeoPixelBus_by_Makuna\src/NeoPixelBus.h:58:0,

                 from C:\Users\Ham\AppData\Local\Temp\arduino_modified_sketch_10798\xdrv_ws2812.ino:31:

C:\Temp\arduino-1.8.0\portable\sketchbook\libraries\NeoPixelBus_by_Makuna\src/internal/NeoEsp8266DmaMethod.h:283:42: note: 'static volatile void NeoEsp8266DmaMethodBase<T_SPEED>::i2s_slc_isr() [with T_SPEED = NeoEsp8266DmaSpeed800Kbps]' was declared here

     volatile static void ICACHE_RAM_ATTR i2s_slc_isr(void)

                                          ^

exit status 1
void osw_osWatch() causes a section type conflict with static volatile void NeoEsp8266DmaMethodBase<T_SPEED>::i2s_slc_isr() [with T_SPEED = NeoEsp8266DmaSpeed800Kbps]

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

arendst, 2017-04-01:
It's a result of having both NeopixelBus library and my Os_watch routine using ICACHE_RAM_ATTR to execute a piece of code in Iram. Why it fails with NeopixelBus only I don't know but I've also seen it with my Pow implementation code where I disable Iram usage if someone wants to use NeopixelBus DMA.

I can't explain what leads to the fault as I'm sure the Arduino-ESP8266 library uses ICACHE_RAM_ATTR too and that doesn't resolves to the error you noticed. I would expect Neopixelbus is the culprit but if someone can resolve it in my code please let me know.

In your case you could remove the usage of ICACHE_RAM_ATTR in osw_osWatch (line 37 in support.ino) and see if it negatively influences you DMA experience.

hamwong, 2017-04-01:
simply avoid **void ICACHE_RAM_ATTR osw_osWatch(void)** is not working,
error pop out realte to line 38

```
Arduino: 1.8.0 (Windows 7), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 1M (64K SPIFFS), ck, Disabled, None"

support:38: error: expected unqualified-id before '{' token

 {

 ^

In file included from C:\Temp\arduino-1.8.0\portable\sketchbook\libraries\NeoPixelBus_by_Makuna\src/NeoPixelBus.h:51:0,

                 from C:\Users\Ham\AppData\Local\Temp\arduino_modified_sketch_32290\xdrv_ws2812.ino:31:

C:\Temp\arduino-1.8.0\portable\sketchbook\libraries\NeoPixelBus_by_Makuna\src/internal/NeoBitmapFile.h:30:22: error: expected declaration before end of line

 #pragma pack(push, 2)

                      ^

exit status 1
expected unqualified-id before '{' token

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

if I void from line 37 - 54 with **//**, I got error as below

```
Arduino: 1.8.0 (Windows 7), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 1M (64K SPIFFS), ck, Disabled, None"

C:\Users\Ham\AppData\Local\Temp\arduino_modified_sketch_603593\support.ino: In function 'void osw_init()':

support:61: error: 'osw_osWatch' was not declared in this scope

   tickerOSWatch.attach_ms(((OSWATCH_RESET_TIME / 3) * 1000), osw_osWatch);

                                                              ^

exit status 1
'osw_osWatch' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

I finally have compile when I // from line 37-54 and line 61, but I don't know will it be issue while operating. I will update here if I have any further founding

arendst, 2017-04-01:
Just replacing
```
void ICACHE_RAM_ATTR osw_osWatch(void)
```
with
```
void osw_osWatch(void)
```
Should have been enough...

hamwong, 2017-04-01:
Thanks @arendst,

it works as your advise, Thank you.

hamwong, 2017-04-01:
Hi @arendst 
one more question,
in user_config.h, I discover if I disable WS2812, than 2812 won't show in manual, same as USE_I2C
but when I disable //#define USE_DS18x20, I still able to see it in GPIO config page,
does disable/enable only for Onewire library? when disable, only not able to use mulit DS18x20?

arendst, 2017-04-01:
When define USE_DS18x20 is disabled the internal DS18B20 routines takeover and therefore you still see the DS18B20

hamwong, 2017-04-01:
understood, Thanks for explain

closed 2017-04-02 by arendst

Sunrise17, 2017-05-17:
do we need to add 1 after define USE_WS2812_DMA? in the description of the precompiled version, pin number is written GPIO3 but it is RX ( not TX).

arendst, 2017-05-17:
Just remove the two slashes.

Sunrise17, 2017-05-17:
i have uploaded to firmware to sonoff basic, and connected data to GPIO1, it started controlling the ws2812 but when i change the colors from my app (publishing several hex colors) as moving over rainbowcircle, it occurs delay to get the colors from led. I changed data cable to GPIO3 but ws2812 does not get any reponse.

Sunrise17, 2017-05-17:
and only first 15leds are getting data and giving response, rests are stuck with one color. I am switching off and on Sonoff, only first 15leds are becoming off and on