arendst/Tasmota #314: Sonoff Basic with BH1750-Sensor

JoeWiseman, 2017-03-31:
Hi,
I just tried to connect a BH1750 amient light sensor to the Sonoff Basic module, but when reading the status 10 I just get the current time ....
OUTPUT:
STATUS10 = {"StatusSNS":{"Time":"2017-03-31T19:30:21"}}

I've connected the 
BH1750   Sonoff
VCC          3V 
GND         GND
SCL           GPIO14
SDA          RX (GPIO3)
like shown in the picture and contigured it like shown in the screenshot ...
[image]
[image]

Would be great if somebody can help me out!
Thanks in Advance
Joe

davidelang, 2017-03-31:
did you do a full poweroff (wait several seconds), restart sequence after the 
config change?

hamwong, 2017-03-31:
did you connect ADDR pin to GND? in my using experience, direct solder will much more stable, without connect mostly failure

ionciubotaru, 2017-04-01:
I2C need pull-up resistors on both SDA and SCL - I don't see it on your picture

arendst, 2017-04-01:
It works flawlessly on a wemos D1 mini...

My sonoffs do not have the GPIO14 but using GPIO01 and GPIO03 I was able to get the BH1750 working:
[image]

[image]

I guess you need a 10k pull up resistor on gpio14 to make it work also.

JoeWiseman, 2017-04-01:
Hi there,

thanks a lot to all of you for answering so fast!!! You're really great!

@davidelang Yes I did a long (over night) break before testing again...

@hamwong I've red that the ADDR could be left unconnected ... If I can't do any damage when I bridge the GND and the ADDR I'm more than willing to try this...

@ionciubotaru I thought that it is possible to connect it directly. Which pull-up resistors (any Link, or description) do I have to take than? Or is it ok to take a 10k Ohm (like arendst posted) on both SDA- & SCL-connector?

@arendst I did not find a clearly description where to find GPIO 03 / 04 / 01 on the board... There are a lot of Pins (5 vertically and 9 horizontally). So what I've found on the net was that I can take a GPIO 03 instead of an GPIO 04 (whereever this pin is to be found) AND that the GPIO 03 is the same as the RX-Pin on the 5er-pin-row. As you call it "Serial In" in your firmware I guess that the GPIO 01 / "Serial Out" ist the TX-pin???? Am I right? Though I don't want to cause any damage to the board I want to be sure before trying.

At least my favorite solution would be the one, given be arendst, because it seem to come along wihtout any resistors.

Regards
Joe

ionciubotaru, 2017-04-01:
Details about I2C pull-up resistors [http://electronics.stackexchange.com/questions/102611/what-happens-if-i-omit-the-pullup-resistors-on-i2c-lines](http://electronics.stackexchange.com/questions/102611/what-happens-if-i-omit-the-pullup-resistors-on-i2c-lines)

arendst, 2017-04-01:
As you found out yourself gpio1 is just fine.

hamwong, 2017-04-01:
> I don't want to cause any damage to the board 

I think GY-302 not that easy to get damage, I was reverse plug the VCC and GND for few minutes until I found it's not running and PCB get hot, but it's still  run fine when plug back to where it should be.

I didn't put resistors, but before I solder the cable, it somehow not very stable, sometimes work, sometimes not, until I solder a cable from ADDR to GND and it works every time I turn on / restart ESP.

now I got 3 ESP with GY-302 and all working reliable without resistors

JoeWiseman, 2017-04-01:
Yeah! Great .... Connecting the sensor to the GPIO 03 / (TX on the board) instead of using GPIO 14 worked fine, just wihtout a brigde between GND and ADDR on the sensor board!
Thanks a lot for your help !!!!
This topic can be closed!

closed 2017-04-02 by arendst

michad2, 2017-12-30:
works fine on sonoff th16, the only problem is the german language. In german the readings name is "Beleuchtungsintensität", node-red have problems with "ä", the json extract node is marked with error

arendst, 2017-12-30:
See #1473