arendst/Tasmota #315: several delay for external buton when internet lost

solucionesmipc, 2017-04-01:
hi:

i connected external button to gpio14 and ground, 
if internet is lost then actions from external button takes several time to execute

is possible to change order of somehing in code to execute action inmediately with or without internet?

thanks Diego.

ionciubotaru, 2017-04-01:
This happens because MQTT tries to reconnect. Please disable all internet service (MQTT, RSYSLOG, timeserver) and you will not see delays any more.

solucionesmipc, 2017-04-01:
i understand it

is there some intermediate solution , i want to keep mqtt functionality

thanks for your answer.

ionciubotaru, 2017-04-01:
no, you have to manually choose between mqtt and no mqtt
better use a local mqtt server, is faster, more stable and you can use it with no encryption

jonathanmbradshaw, 2017-04-01:
if you need to export the switch status to an external MQTT instance you might consider using an MQTT bridge from the local to the remote.

davidelang, 2017-04-02:
the delay you are complaining about is due to a bug in a library that is used by tasmota, when it initiates a connection to the mqtt server, it can do nothing else until it either connects or times out.

run a local mqtt server or disable mqtt to avoid this problem

closed 2017-04-02 by davidelang

solucionesmipc, 2017-04-03:
hi

thnks for your answer, do you know some similar project without that bug?

thanks again

davidelang, 2017-04-03:
it would have to be one that doesn't use the standard Arduino based TCP 
libraries.

someone mentioned some async TCP library, but I don't know any details around 
that.

if you are not using mqtt, disable it. If you are, have it connect to a local 
server so you aren't dependent on the Internet.

solucionesmipc, 2017-04-03:
do you believe can i change the time for timeout?

what file should i modify?