arendst/Tasmota #317: Flashing firmware with  FTDI-programmer with OVERWRITING EVERYTHING

JoeWiseman, 2017-04-01:
Hi there,

I've got a Sonoff basic module whose firmware seems to be broken, because it does not connect to my network anymore and even the button-function doesn't seem to work anymore.....

Now I thought that it would be the best way just to flash the module again, but it seems to be that some wrong settings (or whatever keeps the module from working) is not been overwritten.
For flashing the software I use a FTDI-programmer along with the Arduino-software (but I just use it as described in the tutorial, so I'm really not familiar in using it!!!)

Can anyone out there tell me how to be VERY sure that all setting / programms are erased before flashing again?

Thanks in advance
Joe

arendst, 2017-04-01:
Change define CFG_HOLDER to another value; re-compile and upload to sonoff.

JoeWiseman, 2017-04-01:
Any other value e.g. from 0x20161209 to 0x20161210 will do?

davidelang, 2017-04-01:
yes, read the troubleshooting page on the wiki

JoeWiseman, 2017-04-01:
THX, I'll try it!

JoeWiseman, 2017-04-01:
Great to have you by my side ;-) 
Everything worked out fine.... the module is working again ...
When testing the module I figured out that the WiFi-connection for this specific module is very weak. So when writing the configuration in front of my laptop (where I am sitting in between the Router in the next room and the module) it seems to loose the connection and therefore gets an error while writing. When putting the module somewhere else (closer to the router) everyting works fine.

So thanks a lot as always!!
This topic can be closed!

closed 2017-04-02 by arendst