arendst/Tasmota #318: Change GPIO Pin assignments for NODEMCU and toggle relay for only 2 seconds

citrusfizz, 2017-04-01:
I need a bit of help,  
i'm using a nodeMCU module 
i have 3 items attached,  a DHT22 sensor, a toggle switch, and the "relay"

the DHT22 is on gpio 5
the switch is on gpio 4 (which doesn't really need a change since you expose that one)
and the "relay" which is actually a mosfet, is on gpio 15

i need some help to change these values in the code,  
also,  i need the "relay"  to just toggle on for 2 seconds and then turn off,  as opposed to staying on (like normal until something turns it off again)

thanks for any help!

arendst, 2017-04-01:
Read the wiki. For all your wishes no code changes are needed. It's all up to you.

citrusfizz, 2017-04-01:
Should have guessed,  i haven't been keeping up with the new stuff in the code for a while. 

if anyone else wants to know, i'm making a garage door opener with Temp reporting. 

I set the device in the config to be a Wemos D1 Mini
changing it to this, exposes all of the GPIO pins in the config menu

i set my pins to use  Button1, DHT22, Relay1

cmnd/garagedoor/PulseTime1  5  <---  since the relay is actually a logic level mosfet, this "connects" the two wires to my garage door for half a second

cmnd/garagedoor/PowerOnState 0  <--  this insures that it doesn't toggle the relay if the device reboots

Funny thing is i spent the last two days writing my own code to do this,  and got stuck with the Async-MQTT library  not behaving well in loops and delays().  the trying to convert floats to strings and trying to convert and Char pointers to strings and got frustrated, didn't occur until today that i might be able to use Tasmota

@arendst   you can close this out.   thanks for pointing me in the right direction

closed 2017-04-02 by arendst