arendst/Tasmota #319: i2c LCD display for weather data

MacWyznawca, 2017-04-01:
I added a function to display the temperature (humidity and pressure depending on the connected sensors) using the display i2c LCD 16x2 characters.

All add-ons to the code are labeled USE_I2CLCD 
I used the library: LiquidCrystal_I2C.h

I2c display address: #define I2CLCDADDR 0x3f

Maybe someone will use it, and maybe Theo will add to the project 😉

[Source code **Sonoff-Tasmota-master-4.1.1-I2CLCD.zip**](http://macwyznawca.pl/tasmota/Sonoff-Tasmota-master-4.1.1-I2CLCD.zip)

[image]

[image]

[image]

davidelang, 2017-04-01:
how much does enabling this add to the flash/ram size of the image?

MacWyznawca, 2017-04-01:
@davidelang 
RAM ~1KB

Program: ~3KB

Now I have:

Program size	433kB

Free program space	504kB

Free memory	27kB

MacWyznawca, 2017-04-01:
@davidelang 
Exactly: 430809  - 429161 = 1648

MacWyznawca, 2017-04-02:
Maybe @arendst added this function optionally "quoted by default"? 😉
// #define USE_I2CLCD

arendst, 2017-04-02:
I'll look into it...

smadds, 2017-04-03:
If you are thinking about it, Theo, it would be good if the code was flexible enough to support other displays. Little OLEDS like [this](http://www.ebay.co.uk/itm/3-5V-0-96-12C-Serial-128X64-OLED-LCD-LED-Display-Module-for-Arduino-White-UQ/272555671107) are very useful and small enough to fit in the top of a Sonoff case. There again, I realise how tight space is...
[image]

MacWyznawca, 2017-04-03:
@smadds 🙂
I have scheduled for delivery TODAY by DGL this 😉 128x64 i2c, but with this dusplay is more to do. Ex. character set (fonts) and pictograms.
I try this by weekend 😉

decologne, 2017-04-13:
It's a good idea to extent this nice tool with a small display but make it universal to implement only a driver and let the main controller, openHAB etc, format the information to display. Then it is possible to display information of different sources. Also using of 2 or 4 lines is possible.

MacWyznawca, 2017-04-13:
@smadds @decologne
I spend lot of time for oled and other oled library. Don't good work with Tasmota. Always I have some problems with glitches on display. Displays work good with other code. 
For Tasmota only char LCD works good 😉 

Oled I implemented remotely on other device. I collect data via MQTT from Tasmota and displaying this. Work in progress 😉

petrfaitl, 2017-06-15:
@MacWyznawca Have you made any progress on the OLED support? Or is it best to go with a second controller, picking up MQTT? Thanks

eazuka, 2018-05-16:
??

ascillato2, 2018-07-10:
Hi,

Now available the experimental display branch. Closing this issue. Please reopen if needed. Thanks.

closed 2018-07-10 by ascillato2