arendst/Tasmota #320: 4.1.1 rebooting when selecting AP mode on Wemos

smadds, 2017-04-03:
I have loaded the generic 4.1.1 onto a Wemos D1, but am unable to enter access point mode. Looking at the serial output the other modes seem to work (smartconfig & wps) but not AP mode.

The following scrolls through (along with reams of debug):

00:00:11 RSLT: stat/sonoff/RESULT = {"WifiConfig":"Wifimanager selected"}
00:00:12 Wifimanager: Active for 1 minute

Exception (29):
epc1=0x4000e1c3 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000018 depc=0x00000000

ctx: cont 
sp: 3fff2190 end: 3fff26b0 offset: 01a0

Any ideas?

arendst, 2017-04-03:
Try to enable seriallog 3 and see how far it gets. Just tested it myself but it doesn't fail on me. Do you see more stack output with the exception?

smadds, 2017-04-03:
I can't seem to get it to respond to seriallog 3

I'm wondering if it's something to do with the way I am programming it? I was finding that compiling the code using Arduino IDE then uploading it via serial was working fine, but at the first OTA update it cleared all the wifi details. in order to be sure I had the same code as the OTA (which is the .bin file from Github) I have tried using esptool (under Windows):
```
esptool.exe -vv -cd nodemcu -cb 921600 -cp COM6 -ca 0x00000 -cf "C:\Users\Simon\Google Drive\Arduino\Sonoff-Tasmota\api\arduino\sonoff.ino.bin"
```
It seems to program with no errors and boots as you would expect.

Here's the full log of it cycling, even if I do not try to put it into AP mode:
```
00:00:00 APP: Project sonoff Sonoff (Topic sonoff, Fallback DVES_1957EB, GroupTopic sonoffs) Version 4.1.1
00:00:00 Wifi: Connecting to AP2 indebuurt2 in mode 11N as sonoff-6123...
00:00:07 Wifi: Connect failed as AP cannot be reached
00:00:07 Wifi: Connecting to AP1 indebuurt1 in mode 11N as sonoff-6123...
00:00:14 Wifi: Connect failed as AP cannot be reached
00:00:14 Wifimanager: Active for 1 minute

Exception (29):
epc1=0x4000e1c3 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000018 depc=0x00000000

ctx: cont 
sp: 3fff2150 end: 3fff26b0 offset: 01a0

>>>stack>>>
3fff22f0:  40259b7d 00000017 60000200 40107ca0  
3fff2300:  4022cda1 402564ea 00000001 40225cd2  
3fff2310:  504f550c 40228297 3fff391c 3fff3aa4  
3fff2320:  4022cfe2 3fff391c 3fff3aa4 3fff1490  
3fff2330:  00000000 3ffef110 3ffef08c 3fff3aa4  
3fff2340:  3fff1490 3fff1490 00000000 3fff1490  
3fff2350:  00000018 00000064 19a620a2 fffeffff  
3fff2360:  3ff20a00 0000ffff 3fff2388 4022893a  
3fff2370:  3ffef08c 3fff3aa4 3fff3aa4 3fff1490  
3fff2380:  19a620a2 3fffeb57 00000000 00000000  
3fff2390:  00000000 00000000 00000000 00000000  
3fff23a0:  00000000 00000000 00000000 00000000  
3fff23b0:  00000000 00000000 00000000 40229a8c  
3fff23c0:  3fff3aa4 00000000 00000000 40229ad4  
3fff23d0:  00000000 3ffef110 3ffefe98 00000000  
3fff23e0:  402394a9 00000002 696d2031 6574756e  
3fff23f0:  402395f9 00000002 00000002 3fff1490  
3fff2400:  3ffef284 40239682 00000002 00000000  
3fff2410:  4021795f 402180c4 64656863 3ffe8500  
3fff2420:  3fff132c 3fff0a15 3fff1490 4020e73d  
3fff2430:  69666957 6843203a 696b6365 6320676e  
3fff2440:  40239442 3fff0a14 3fff1490 3fff0a13  
3fff2450:  3fff1490 3fff0a10 00000002 4020e867  
3fff2460:  3fff34e4 402180c4 0000000c 3fff0a15  
3fff2470:  3fff0a13 3fff1490 3fff0a14 4020e9d6  
3fff2480:  00000000 40101d53 00000000 3ffeee20  
3fff2490:  00000005 00000000 00000020 401023f2  
3fff24a0:  3ffeee50 40101f86 00000001 3ffe855c  
3fff24b0:  3fff132c 3fff0a15 3fff0a10 4020eb30  
3fff24c0:  00000001 00000000 00000002 401023f2  
3fff24d0:  401056b4 3fff394c 3ffeee20 401023f2  
3fff24e0:  00000001 401048c3 3ffeee50 401023f2  
3fff24f0:  40104d26 401056cf 3ffee760 40103294  
3fff2500:  40103105 3ffee760 3ffef110 40103294  
3fff2510:  4010480a 0000007f 7fffffff 00000000  
3fff2520:  0000007f 00000000 00000000 00000100  
3fff2530:  4010376f 00080000 7fffffff 00000000  
3fff2540:  00000000 40104639 00000000 401078d5  
3fff2550:  3fffc278 00000005 3fff0f2c 3ffe855c  
3fff2560:  3fff132c 00000005 3fff0e24 4021366e  
3fff2570:  4010731d 00000030 00000012 ffffffff  
3fff2580:  40107c68 0000077e 00001f40 00000003  
3fff2590:  00000008 00000914 00000000 fffffffe  
3fff25a0:  ffffffff 3fffc6fc 00000001 00000000  
3fff25b0:  00000000 3fffdad0 3fff1688 00000030  
3fff25c0:  ffffffff 3fffc6fc 00000038 3fff132c  
[... 74 lines cut ...]
6f 6e 31 00 53 65 72 69 61 6c 20 4f 75 74 00 00 53 65 72 69 61 6c 20 49 6e 00 00 00 52 65 6c 61 
79 31 00 00 4c 65 64 31 49 00 00 00 53 65 6e 73 6f 72 00 00 7b 76 7d 00 43 6f 6e 66 69 67 20 6d 
6f 64 75 6c 65 00 00 00 7b 6d 74 7d 00 00 00 00 3c 2f 73 63 72 69 70 74 3e 00 00 00 3c 62 6f 64 
79 20 6f 6e 6c 6f 61 64 3d 27 73 6c 28 29 27 3e 00 00 00 00 20 73 65 6c 65 63 74 65 64 20 00 00 
43 6f 6e 66 69 67 20 6c 6f 67 67 69 6e 67 00 00 7b 62 30 7d 00 00 00 00 7b 62 31 7d 00 00 00 00 
4c 4f 47 5f 4c 45 56 45 4c 5f 49 4e 46 4f 00 00 7b 62 32 7d 00 00 00 00 6c 73 00 00 7b 61 00 00 
6c 77 00 00 4c 4f 47 5f 4c 45 56 45 4c 5f 4e 4f 4e 45 00 00 6c 6c 00 00 7b 6c 32 7d 00 00 00 00 
7b 6c 33 7d 00 00 00 00 7b 6c 34 7d 00 00 00 00 20 63 68 65 63 6b 65 64 00 00 00 00 43 6f 6e 66 
69 67 75 72 65 20 4f 74 68 65 72 00 7b 70 31 7d 00 00 00 00 7b 72 31 7d 00 00 00 00 7b 31 00 00 
7b 32 00 00 7b 33 00 00 7b 72 32 7d 00 00 00 00 7b 72 33 7d 00 00 00 00 7b 72 34 7d 00 00 00 00 
49 6e 66 6f 72 6d 61 74 69 6f 6e 00 31 2e 35 2e 33 00 00 00 57 69 66 69 3a 20 50 61 74 63 68 20 
69 73 73 75 65 20 32 31 38 36 00 00 20 42 47 4e 00 00 00 00 2e 6c 6f 63 61 6c 00 00 2f 63 6e 00 
2f 6d 64 00 2f 77 31 00 2f 77 30 00 2f 6d 71 00 2f 64 6d 00 2f 6c 67 00 2f 63 6f 00 2f 64 6c 00 
2f 73 76 00 2f 72 73 00 2f 72 74 00 2f 75 70 00 2f 75 31 00 2f 75 32 00 2f 63 6d 00 2f 63 73 00 
2f 61 78 00 2f 61 79 00 2f 72 62 00 2f 66 77 6c 69 6e 6b 00 2f 75 70 6e 70 2f 63 6f 6e 74 72 6f 
6c 2f 62 61 73 69 63 65 76 65 6e 74 31 00 00 00 2f 65 76 65 6e 74 73 65 72 76 69 63 65 2e 78 6d 
6c 00 00 00 2f 73 65 74 75 70 2e 78 6d 6c 00 00 2f 64 65 73 63 72 69 70 74 69 6f 6e 2e 78 6d 6c 
00 00 00 00 2a 00 00 00 49 6e 69 74 69 61 6c 69 7a 65 64 00 68 74 74 70 00 00 00 00 69 64 78 00 
6e 76 61 6c 75 65 00 00 73 76 61 6c 75 65 31 00 69 74 00 00 6f 74 00 00 2f 69 6e 76 61 6c 69 64 
2f 00 00 00 2f 63 6f 6e 66 69 67 00 2f 67 72 6f 75 70 73 00 2f 73 63 68 65 64 75 6c 65 73 00 00 
2f 73 65 6e 73 6f 72 73 00 00 00 00 2f 73 63 65 6e 65 73 00 2f 72 75 6c 65 73 00 00 42 48 31 37 
35 30 00 00 20 6d 53 00 41 44 4d 49 4e 00 00 00 3f 00 00 00 46 41 4c 53 45 00 00 00 53 54 4f 50 
00 00 00 00 54 52 55 45 00 00 00 00 53 54 41 52 54 00 00 00 42 4c 49 4e 4b 00 00 00 42 4c 49 4e 
4b 4f 46 46 00 00 00 00 53 54 41 54 55 53 00 00 50 4f 57 45 52 4f 4e 5
```

arendst, 2017-04-03:
If connected serially you should be able to execute command seriallog 3 during the first seconds before the exception.

It probably clears your settings as the CFG_HOLDER in your source is different from the one on Github.

You say you do not put it in AP mode but as the text says: ``00:00:14 Wifimanager: Active for 1 minute`` it is starting in AP mode as a result from the flash config (wificonfig 2).

It looks to complain about webserver port. Did you change the port from default 80?

smadds, 2017-04-03:
This is the .bin file from Github, so no I have not changed anything. I've done a 4sec "clear all" to ensure nothing is stored locally.

I'll try again on a virgin board using the esptool and let you know how I get on.

smadds, 2017-04-03:
I think I have found the problem. When programming the Wemos using Arduino IDE I was selecting board type NodeMCU 1.0. This programmed fine, but didn't cope with OTA reprogramming. By doing the original programming as a Generic ESP8266  1M/64SPIFFS the device programs and then the OTA is fine. Happy to close this ticket, but worth noting for other non-Sonoff users

closed 2017-04-03 by smadds