arendst/Tasmota #321: IRhvac not working

smadds, 2017-04-03:
Delighted to have this in the code, but can't seem to get it working. Not sure if it's me being stupid or a bug!

I have defined GPIO15 as "08 IRRemote", but when I issue the command in the console I get a Command Unknown error:

`
21:49:18 CMND: IRhvac {"HVAC_Mode": "HVAC_HOT" , "HVAC_FanMode" : "FAN_SPEED_AUTO" , "HVAC_Temp" : 24 , "HVAC_Vendor" : "Mitsubishi" }
`
`
21:49:18 MQTT: stat/sens-lobby01/RESULT = {"Command":"Unknown"}
`

arendst, 2017-04-03:
That's probably because irhvac is disabled in the distribution because of max ota memory available.

Just enable it in user_config.h and recompile. Make sure to use the correct irremoteesp8266 library as noted in the wiki.

Once uploaded you may use the intermediate minimal version to reload your next production one as the ota memory space is too small to do it at once.

smadds, 2017-04-03:
thanks Theo.

Re options - yes, I copied the command from the original ticket and hadn't checked for syntax changes.

closed 2017-04-03 by smadds