arendst/Tasmota #322: Problems setting hostname

smadds, 2017-04-04:
I'm trying to set the hostnames to just "%s", without the partial MAC. If I try from the web interface or console the command appears to work, but returns the original hostname:
`08:46:26 CMND: hostname %s`
`08:46:26 MQTT: stat/sens-lobby01/RESULT = {"Hostname":"%s-%04d"}`

Also, if I send a group command via MQTT:
`cmnd/sonoffs/hostname %s`
I get responses like:
`stat/s20-kitchen-dehumid/RESULT {"Command":"Unknown"}`

Are these expected behaviours?

khcnz, 2017-04-04:
Hi @smadds unfortunately the only supported options are the default one (with %s-%04d) or  a completely custom one without %s. The code checks for %s and if it is in the string uses the default one. Safe string formatting in c is hard with formatters so a failsafe approach has been taken

khcnz, 2017-04-04:
I have added a note here https://github.com/arendst/Sonoff-Tasmota/wiki/Commands under the hostname command

smadds, 2017-04-04:
OK, thanks for getting back, @khcnz.

I can see why the group command for hostname would be undesirable!

closed 2017-04-04 by smadds