arendst/Tasmota #323: Can not establish connection TCP / IP at 80 Port

ChienTing, 2017-04-04:
I used the command with "Sonoff" unable to establish the connection TCP / IP at 80 Port
Please help me thank you

[image]

citrusfizz, 2017-04-04:
you need to describe your problem  in way more detail than this.  Where did you enter this "command"
did you load the software on your device?  did you specify your SSID and Pass in the config? Is the device taking an IP via DHCP? 

why did you post that picture?

ChienTing, 2017-04-05:
I brush Tasmota V4.1.0 on Sonoff
Set up also connected to my WiFI
You can also control the switch on the web page

But I need to use the command to go under the command control switch
Such as SSH

I posted this picture to illustrate "Three Way Handshake"
Is not it necessary to establish a connection?

*Normal Connection Establishment: The "Three Way Handshake"*

To establish a connection, each device must send a *SYN* and receive an
*ACK* for it from the other device. Thus, conceptually, we need to have
four control messages pass between the devices. However, it's inefficient
to send a *SYN* and an *ACK* in separate messages when one could
communicate both simultaneously. Thus, in the normal sequence of events in
connection establishment, one of the *SYNs* and one of the *ACKs* is sent
together by setting both of the relevant bits (a message sometimes called a
*SYN+ACK*). This makes a total of three messages, and for this reason the
connection procedure is called a *three-way handshake*.

http://www.tcpipguide.com/free/t_TCPConnectionEstablishmentProcessTheThreeWayHandsh-3.htm

The attachment is my network test software

Thanks ...

​
 USR-TCP232-Test-V1.3.zip
<https://drive.google.com/file/d/0B1kXx8Diw4pOOF8yVXZTUWxuUUk/view?usp=drive_web>
​

davidelang, 2017-04-05:
yes a 3 way handshake is part of establishing a connection, but you have shown nothing to indicate that the handshake is failing.

back up a bit.

1. can you ping the device

2. when you try and connect to port 80, what happens? (and what tool do you use to try and connect)

3. do you have the ability to get logs from this device (see the troubleshooting page of the wiki)

I still don't see any case where posting a picture of a three way handshake helps explain what's failing.

ChienTing, 2017-04-05:
Thank you for your reply

1.   Yes i can ping the device OK

2.   I can connect to port 80 but can only connect with web pages

davidelang, 2017-04-05:
Ok, port 80 is supposed to be the web pages, what are you expecting to get there 
instead of web pages?

ChienTing, 2017-04-05:
I would like to be able to use the command to send the way to control the
switch instead of the web page under the instructions
If you can use my attachment network test software to send orders it would
be great

Thanks

davidelang, 2017-04-05:
If you want to control the sonoff via the tasmota software, you can use mqtt, or 
the hue/wemo emulation to do so.

If you want to create an additional http based method of controlling the device, 
that would be a major enhancement (unlikely to happen due to resource 
contraints) and you may be better of using some other firmware as a base.

But this doesn't match the issue title of "can not establish connection", you 
are establishing a connection just fine, you just wish there was functionality 
available that isn't there.
k

ChienTing, 2017-04-05:
Ok! I look for other firmware
thank you very much

ChienTing, 2017-04-13:
I find the other firmware or the same can not connect
Is there any way?
I need to use ssh with 22 Port
Go under "command"
Thank you

2017-04-05 15:44 GMT+08:00 ting <ting1688@gmail.com>:

davidelang, 2017-04-13:
the Tasmota software does not support ssh, you can issue commands via the serial 
port, the web console, or via mqtt

On Wed, 12 Apr 2017, ChienTing wrote:

> I find the other firmware or the same can not connect
> Is there any way?
> I need to use ssh with 22 Port
> Go under "command"
> Thank you
>
> 2017-04-05 15:44 GMT+08:00 ting <ting1688@gmail.com>:

ChienTing, 2017-04-13:
Ok  !

thank you very much

closed 2017-04-23 by davidelang