arendst/Tasmota #324: Sonoff stalling at regular intervals

Courty40, 2017-04-04:
System was working on 4.0.8 but on 4.1.1 it keeps stalling for a few seconds and this appears on the log.
Looks like its trying to auto find the MQTT servers even though its attached to a MQTT server and its working.
This is running on a TH10

15:29:22 MQTT: Attempting connection...
15:29:23 mDNS: Query done with 0 mqtt services found
15:29:23 MQTT: Connected
15:29:23 MQTT: tele/Node8T/LWT = Online (retained)
15:29:23 MQTT: cmnd/Node8T/POWER = 
15:29:24 UPnP: Multicast (re)joined

Courty40, 2017-04-04:
//#define USE_DISCOVERY stops the stalling so issue looks to be with the MQTT Discover

davidelang, 2017-04-23:
There is a known problem with the TCP libraries that cause everything to stall when trying to establish a connection out (like to a mqtt server). please increase your debugging level so we can see exactly what's happening (see the troubleshooting page)

closed 2018-05-08 by stale[bot]