arendst/Tasmota #325: Sonoff Touch as a trigger

ageorgios, 2017-04-04:
Hello,

Can we use sonoff touch wall button as a button trigger for home automation and not only for operating the switch that the hardware is on?

Thank you.

davidelang, 2017-04-04:
yes, look at buttontopic if you want it to send a command rather than switch the 
local hardware.
k

ageorgios, 2017-04-04:
can it be done on a S20 Socket Module also?
where is the buttontopic configuration? is it available on the web interface?

davidelang, 2017-04-04:
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

it can be entered on the web console, I would have to look through all the 
configuration options to see if it's in another menu as well.

this can apply to the button on any sonoff device.

davidelang, 2017-04-23:
closing this issue as it seems the discussion has wound down. please re-open if you still have a problem

closed 2017-04-23 by davidelang