arendst/Tasmota #326: Multiple DS18B20 in Domoticz

lkhaskjdhlkashd, 2017-04-04:
Hi!

First off all, Great work on the Sonoffs!

I wonder if it is possible to have 2 DS18B20 sensors on 1 Sonoff added in Domoticz.
1 Works perfectly, but when i connect another sensor parallel to the first on, it shows up on the webpage en in the log file, but i cant find a way to add the second sensor in Domoticz..

Greetings
-Ikhas

davidelang, 2017-04-23:
that sounds like a Domoticz issue. the tasmota firmware sends all the sensor output in one json message, so your home automation software would need to look for the second temp sensor.

I'm going to go ahead and close this as it doesn't look like a tasmota issue. If you disagree please re-open the issue (and you may want to update the wiki documentation for Domoticz when you figure this out)

closed 2017-04-23 by davidelang

Bodzio122333, 2017-09-11:
Hello,
I connected two ds18b20 to Wamos D1 mini but I see one idx temperature in Configuration Domiticz menu. So module send one temprature to Domoticz. Please give me information how add two ds18b20 to domoticz.

[image]
[image]