arendst/Tasmota #327: Pair RF with Slampher after Tasmota Is flashed?

citrusfizz, 2017-04-04:
Is there a way to pair an rf remote with the slampher modules after tasmota has been flashed?
to flash, i had to ground R20 while reseting power,   but it doesn't appear i can put it into program mode after the new firmware flash

thanks 

BTW  @arendst  I know that GPIO1 (serial out) works with my dht sensor   not sure why the slampher config doesn't expose the GPIO

arendst, 2017-04-08:
Just follow the text from the itead website regarding pairing your 434MHz remote:

_Quickly press "SET" button on Sonoff and Slampher twice, the red LED on Sonoff/Slampher will flash one time, then
press one of the ABCD buttons on the controller once for a few seconds for pairing. If you pair it with A, then you can
press A to turn on/off your device.
(http://blog.iteadstudio.com/wp-content/uploads/2015/12/sonoff_rf_clear_pairing_with_433.gif)
To cancel the pairing, you need to fast press the"SET" button on Sonoff/Slampher 3 times. If you short press the
“SET” button, it will become a switch button, with it you can turn connected appliances on or off._
It works fine with the itead provided remote control (just tested on the Sonoff RF). It doesn't work with my KaKu remote control.

citrusfizz, 2017-04-08:
worked when screwed into a socket, i was trying that when i was powering over serial. 

Thanks for inspiring me to try again

closed 2017-04-10 by arendst