arendst/Tasmota #329: Enhancement reqest: Show AP Mac address

stefanbode, 2017-04-04:
Small request to add the AccessPoint Max in the feedback of:

void state_mqttPresent(char* svalue, uint16_t ssvalue)

  String routerMAC;
  routerMAC =  WiFi.BSSIDstr();
  snprintf_P(svalue, ssvalue, PSTR("%s, \"Wifi\":{\"AP\":%d, \"SSID\":\"%s\", \"RSSI\":%d, \"ApMAC\":\"%s"}}"),
    svalue, sysCfg.sta_active +1, sysCfg.sta_ssid[sysCfg.sta_active], WIFI_getRSSIasQuality(WiFi.RSSI()), routerMAC.c_str() );
  
With this enhancement I can check to which of my AP the thing is finally connected. They all have the same SSID. Therefore the MAC is the only working identifier.

davidelang, 2017-04-04:
look at the output of status6 it includes client, which contains the last 6 of 
the mac address (and since the first 6 will be the same for every sonoff, this 
should be sufficient)

stefanbode, 2017-04-05:
Hi David, this is not what I was requested. The clientID is fine and need no extra implementation. The function WiFi.BSSIDstr(); give you the MAC address of the Wifi Access Point the client is connected to. This is something different. I have 6 Accesspoint and therefore it is sometimes not easy to understand why RSSI is that low, if you don't know, which AP is really connected. This information is currently not included.

ionciubotaru, 2017-04-05:
Just add a pull request and Theo Arendst will decide or not to include it in the next release

arendst, 2017-04-08:
Implemented in next release

closed 2017-04-29 by arendst