arendst/Tasmota #330: software serial

aferraz, 2017-04-05:
@arendst, it's possible to add the ability to read from a serial port (software serial) ?
For example use the sonoff or electrodragon to read from a modbus device like a Power Energy Meter

Thanks
 Alessandro

davidelang, 2017-04-05:
the sonoff dual does use the serial port, so it is possible. The question is 
going to be how useful the code would be vs the size of the code. If it's a 
common device, it could be arguably useful enough to add the code to the common 
image.

If it's a less common device, it may still be possible to include the code 
contributed by someone if it's disabled by default (with an ifdef like other 
things are)
k

davidelang, 2017-04-23:
closing this issue as there has been no response. please re-open if you think this needs more discussion.

closed 2017-04-23 by davidelang