arendst/Tasmota #331: Sonoff Touch respons Unknown

smekcio, 2017-04-05:
I have issue that it doesnt matter what command I send I always receive {"Command":"Unknown"} on topic stat/.../RESULT

davidelang, 2017-04-05:
what commands have you attempted to send, how are you sending them (serial, web 
console, mqtt??)

smekcio, 2017-04-05:
Im trying via:
- serial (ex. status/power on)
- web htttp (address/?cm=status)
- console (status/power on)
And tej most value for me mqtt by publishing on topics:
cmnd/sonoff/POWER with message ON or cmnd/sonoff/STATUS and Im always receiving {"Command":"Unknown"} on topic stat/.../RESULT

smekcio, 2017-04-06:
The problem disappeared when I changed MQTT topic from salon/swiatlo (english: livingroom/light) to just "light".

closed 2017-04-06 by smekcio

davidelang, 2017-04-06:
yes, '/' has special meanings in mqtt, so it should nto be used in a topic

flanshaw, 2017-04-10:
I've got the same issue. Pressing the button (for over 4 seconds) to reset no longer work. any advice how i can reset to user_config.h.

davidelang, 2017-04-10:
wee the troubleshooting page for the options to reset the device