arendst/Tasmota #332: Connect to livolo touch panel

j105, 2017-04-05:
Hi,

Any chance you have idea in regards connecting sonoff with livolo touch panel?

Based on research it looks like using an TTP223 touch module for touch ability.

I am more than happy to donate one of these touch switch in order to support.

Cheers
Jimmy Fan

ionciubotaru, 2017-04-05:
Are you sure that Livolo is based on ESP8266 ?
Please send some links to schematic or at least some pictures of inside components.

j105, 2017-04-05:
It's not. I have ordered few sonoff basic for main unit and trying use
touch panel itself as button.

Cheers

davidelang, 2017-04-05:
what is the interface to the touch panel?

j105, 2017-04-06:
[image]
That's what I found. It looks using TTP223 @ionciubotaru @davidelang

davidelang, 2017-04-06:
you misunderstood my question. I don't care how it works internally, how do you talk to it?

does it provide an output signal (0v if nothing is touched, 5v if something is for example)?

or do you talk to it via a uart interface, a spi interface, a i2c interface, something else?

do you have to hook wires to the LED pins to read a signal out of it?

there are a lot of motion sensors that have very sophisticated innards, but their interface is that they close a relay when they detect motion.

That's what I mean when I ask what the interface to it is.

ionciubotaru, 2017-04-06:
The leds are controlled by GPIO 0 and 20 of the main IC (labeled ZUNO)
Connect those pins via  1K serial resistor to GPIO 1 and GPIO 3 (TX and RX) of your sonoff basic) and connect grounds of your both circuits, then declare TX and RX as SW! and SW2 in Tasmota interface, then check the sensor status via MQTT

j105, 2017-04-06:
@davidelang It seems using GPIO for communication. 

Could you please take a look following video.
https://www.youtube.com/watch?v=ykqv4bZ9zb0

TBH, I don't have much experience in terms of EE therefore sorry for any confusion.

j105, 2017-04-06:
@davidelang In terms of LED lights, I believe that I have to wire to somewhere to get the status.

It lights in red when switched on and blue when off.

davidelang, 2017-04-06:
unfortuantly that video doesn't tell us much. what little it does tell us is 
that the display is reading some sort of analog value, not just on/off, but it 
doesn't say anything about how the communications is taking place.

GPIO stands for General Purpose Input Output, i.e. it could be anything

j105, 2017-04-06:
@davidelang https://github.com/RB3rg/Livolo-Mod

Is that something interesting to you?

davidelang, 2017-04-06:
Ok, that says that to use these, you need to first reprogram the 
 livolo, then replace the relay on it, then solder wires to it to 
 communicate with it, and finally program something to talk to it.

 The firmware you are pointing to uses a uart to communicate, so that is 
 going to take custom code to talk to this mod.

 That's a lot of changes, so I somewhat doubt that adding support to 
 talk to a device that requires so much work first is going to be a very 
 high priority. If someone contributes the code for it, and it's disabled 
 by default (because it could conflict with other users of the serial 
 port), then there is a possibility that it would be added.

 David Lang

 On Thu, 06 Apr 2017 04:54:41 -0700, Jimmy Fan wrote:
> @davidelang https://github.com/RB3rg/Livolo-Mod
>
> Is that something interesting to you?

j105, 2017-04-07:
@davidelang I am more than happy to help out with the ESP in terms of creating a generic interface for those external inputs. I have ordered that PIC and looking to create a custom firmware to working on.

Will keep you updated. :)

j105, 2017-04-08:
@davidelang I found a way to reprogram the touch panel to have rs232 ability? Could you please advice me how I would change at sonoff side?

davidelang, 2017-04-08:
you will need to look at the code for the dual module, it uses the rs232 
interface to communicate with the relays. You would need to do something similar 
to communicate with the touch panel, but then take the results and make them be 
switch inputs.

On Sat, 8 Apr 2017, Jimmy Fan wrote:

> @davidelang I found a way to reprogram the touch panel to have rs232 ability? Could you please advice me how I would change at sonoff side?

j105, 2017-04-08:
@davidelang I reckon this is what I want.

https://github.com/arendst/Sonoff-Tasmota/blob/master/sonoff/sonoff.ino#L1959

All I need is to connecting touch panel with RX, 3.3v and Ground PIN?

j105, 2017-04-08:
@davidelang by sending cmd like POWERTOGGLE?

davidelang, 2017-04-09:
no, sending a command like power2 toggle is the command to switch a relay

you will need to add code to talk whatever protocol the livolo touch panel 
talks, and when you get the message from the livolo you will need to execute 
code similar to if switch2 was pressed.

k

davidelang, 2017-04-09:
On Sat, 8 Apr 2017, Jimmy Fan wrote:

> @davidelang I reckon this is what I want.
>
> https://github.com/arendst/Sonoff-Tasmota/blob/master/sonoff/sonoff.ino#L1959
>
> All I need is to connecting touch panel with RX, 3.3v and Ground PIN?

that's only the first step, you probably will need to connect tx as well 
(depending on if the communication to the livolo is one-way or two-way), and 
then you will need to write code on the sonoff side to talk to it.

j105, 2017-04-09:
@davidelang I have got the firmware therefore I can define what outputs from livolo.
I see what you say. thanks

davidelang, 2017-04-23:
I'm going to go ahead and close this. Once you get it working, you can submit your changes with a PR and they can be discussed there.

closed 2017-04-23 by davidelang

radiumx3, 2017-11-13:
@j105 Any news on this project? I'm really interested and would'nt mind dig a bit into the code if needed