arendst/Tasmota #333: IR available?

mihalski, 2017-04-05:
Hi,

I have both:
```
#define USE_IR_REMOTE                            // Send IR remote commands using library IRremoteESP8266 and ArduinoJson (+3k code, 0.3k mem)
#define USE_IR_HVAC                            // Support for HVAC system using IR (+2k code)
```
defined, however there is no option to select an IR transmitter for any GPIO.

Has something with regards to configuration changed?

Regards,
Michal

arendst, 2017-04-06:
Are you sure your recompiled version did really load to your sonoff? When enabling IR_HVAC the code tips the balance and might be to big to upload to your device. In that case you need a two step approach where you first compile and upload the minimal version and then the re-compiled full version.

In my case, as soon as I compile IR_REMOTE in I get the option to select it.

mihalski, 2017-04-06:
It seems to be the right version:
```
Program version	4.1.2
Build Date & Time	2017-04-06T07:08:05
Core/SDK version	2_3_0/1.5.3(aec24ac9)
Uptime	11 Hours
Flash write count	142
Boot count	35
Reset reason	Software/System restart
```
[image]

One thing I just noticed that concerns me is that even though I have the #USE_IR defines uncommented I don't see it linking against the IRremoteESP8266 library. There is no error or warning that I can see while it's compiling.
```
Linking everything together...
"/Users/michal/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-gcc" -g -w -Os -nostdlib -Wl,--no-check-sections -u call_user_start -Wl,-static "-L/Users/michal/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/lib" "-L/Users/michal/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/ld" "-Teagle.flash.4m.ld" -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,register_chipv6_phy  -o "/var/folders/g8/78163lzd5t396zkds3b3_pnh0000gn/T/arduino_build_822789/sonoff.ino.elf" -Wl,--start-group "/var/folders/g8/78163lzd5t396zkds3b3_pnh0000gn/T/arduino_build_822789/sketch/core_esp8266_timer.c.o" "/var/folders/g8/78163lzd5t396zkds3b3_pnh0000gn/T/arduino_build_822789/sketch/core_esp8266_wiring_digital.c.o" "/var/folders/g8/78163lzd5t396zkds3b3_pnh0000gn/T/arduino_build_822789/sketch/core_esp8266_wiring_pwm.c.o" "/var/folders/g8/78163lzd5t396zkds3b3_pnh0000gn/T/arduino_build_822789/sketch/sonoff.ino.cpp.o" "/var/folders/g8/78163lzd5t396zkds3b3_pnh0000gn/T/arduino_build_822789/libraries/PubSubClient/PubSubClient.cpp.o" "/var/folders/g8/78163lzd5t396zkds3b3_pnh0000gn/T/arduino_build_822789/libraries/Ticker/Ticker.cpp.o" "/var/folders/g8/78163lzd5t396zkds3b3_pnh0000gn/T/arduino_build_822789/libraries/ESP8266WiFi/ESP8266WiFi.cpp.o" "/var/folders/g8/78163lzd5t396zkds3b3_pnh0000gn/T/arduino_build_822789/libraries/ESP8266WiFi/ESP8266WiFiAP.cpp.o" "/var/folders/g8/78163lzd5t396zkds3b3_pnh0000gn/T/arduino_build_822789/libraries/ESP8266WiFi/ESP8266WiFiGeneric.cpp.o" "/var/folders/g8/78163lzd5t396zkds3b3_pnh0000gn/T/arduino_build_822789/libraries/ESP8266WiFi/ESP8266WiFiMulti.cpp.o" "/var/folders/g8/78163lzd5t396zkds3b3_pnh0000gn/T/arduino_build_822789/libraries/ESP8266WiFi/ESP8266WiFiSTA.cpp.o" "/var/folders/g8/78163lzd5t396zkds3b3_pnh0000gn/T/arduino_build_822789/libraries/ESP8266WiFi/ESP8266WiFiScan.cpp.o" "/var/folders/g8/78163lzd5t396zkds3b3_pnh0000gn/T/arduino_build_822789/libraries/ESP8266WiFi/WiFiClient.cpp.o" "/var/folders/g8/78163lzd5t396zkds3b3_pnh0000gn/T/arduino_build_822789/libraries/ESP8266WiFi/WiFiClientSecure.cpp.o" "/var/folders/g8/78163lzd5t396zkds3b3_pnh0000gn/T/arduino_build_822789/libraries/ESP8266WiFi/WiFiServer.cpp.o" "/var/folders/g8/78163lzd5t396zkds3b3_pnh0000gn/T/arduino_build_822789/libraries/ESP8266WiFi/WiFiUdp.cpp.o" "/var/folders/g8/78163lzd5t396zkds3b3_pnh0000gn/T/arduino_build_822789/libraries/ESP8266HTTPClient/ESP8266HTTPClient.cpp.o" "/var/folders/g8/78163lzd5t396zkds3b3_pnh0000gn/T/arduino_build_822789/libraries/ESP8266httpUpdate/ESP8266httpUpdate.cpp.o" "/var/folders/g8/78163lzd5t396zkds3b3_pnh0000gn/T/arduino_build_822789/libraries/ESP8266WebServer/ESP8266WebServer.cpp.o" "/var/folders/g8/78163lzd5t396zkds3b3_pnh0000gn/T/arduino_build_822789/libraries/ESP8266WebServer/Parsing.cpp.o" "/var/folders/g8/78163lzd5t396zkds3b3_pnh0000gn/T/arduino_build_822789/libraries/DNSServer/DNSServer.cpp.o" "/var/folders/g8/78163lzd5t396zkds3b3_pnh0000gn/T/arduino_build_822789/arduino.ar" -lm -lgcc -lhal -lphy -lpp -lnet80211 -lwpa -lcrypto -lmain -lwps -laxtls -lsmartconfig -lmesh -lwpa2 -llwip_gcc -lstdc++ -Wl,--end-group  "-L/var/folders/g8/78163lzd5t396zkds3b3_pnh0000gn/T/arduino_build_822789"
"/Users/michal/Library/Arduino15/packages/esp8266/tools/esptool/0.4.9/esptool" -eo "/Users/michal/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/bootloaders/eboot/eboot.elf" -bo "/var/folders/g8/78163lzd5t396zkds3b3_pnh0000gn/T/arduino_build_822789/sonoff.ino.bin" -bm dio -bf 40 -bz 4M -bs .text -bp 4096 -ec -eo "/var/folders/g8/78163lzd5t396zkds3b3_pnh0000gn/T/arduino_build_822789/sonoff.ino.elf" -bs .irom0.text -bs .text -bs .data -bs .rodata -bc -ec
Using library PubSubClient at version 2.6 in folder: /Users/michal/Documents/Arduino/libraries/PubSubClient 
Using library Ticker at version 1.0 in folder: /Users/michal/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/Ticker 
Using library ESP8266WiFi at version 1.0 in folder: /Users/michal/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WiFi 
Using library ESP8266HTTPClient at version 1.1 in folder: /Users/michal/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266HTTPClient 
Using library ESP8266httpUpdate at version 1.1 in folder: /Users/michal/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266httpUpdate 
Using library ArduinoJson at version 5.8.4 in folder: /Users/michal/Documents/Arduino/libraries/ArduinoJson 
Using library ESP8266WebServer at version 1.0 in folder: /Users/michal/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer 
Using library DNSServer at version 1.1.0 in folder: /Users/michal/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/DNSServer 
Sketch uses 415454 bytes (39%) of program storage space. Maximum is 1044464 bytes.
Global variables use 41836 bytes (51%) of dynamic memory, leaving 40084 bytes for local variables. Maximum is 81920 bytes.
```

arendst, 2017-04-06:
Having only 415454 as file size you must have enabled be_minimal as that will disable all defines...

mihalski, 2017-04-06:
Aha! And that is EXACTLY what has happened. Thank you!

Now works like a charm (at least tested with the Toshiba, will test with the Mitsubishi next.

Regards,
Michal

closed 2017-04-06 by mihalski