arendst/Tasmota #334: DS18B20 Negative Temperatures

ThomDietrich, 2017-04-05:
This might be related to #165 and to the fact that I'm using multiple sensor, hence activated the USE_DS18x20 define.

As soon as the temperature for a sensor drops below 0°C I'm seeing readings in the area of: 
```json
{
  "Time":"2017-04-05T23:39:51",
  "DS18x20":{
    "DS1":{"Type":"DS18B20", "Address":"28D025FB05000020", "Temperature":3.8},
    "DS2":{"Type":"DS18B20", "Address":"28FF4E63801603DD", "Temperature":4085.3}
  }
}
```

arendst, 2017-04-08:
I think the following code should solve it. Could you give it a try by changing line 157 in xsns_ds18x20 from
```
      t = temp12 * 0.0625;

```
into
```
      uint16_t temp12 = (data[1] << 8) + data[0];
      if (temp12 > 2047) {
        temp12 = (~temp12) +1;
        sign = -1;
      }
      t = sign * temp12 * 0.0625;
```
and let me know the results please.

ThomDietrich, 2017-04-09:
Hey @arendst, thanks for looking into this. Nice approach.
Is the 0.0625 a micro-optimization you think will make a difference?

I had a quick encounter with #336, then everything worked as expected. I am not able to confirm the reported temperature exactly but I had a look at the datasheet and everything seems fine 😉 

Code working, can be implemented. Thanks as always!

closed 2017-04-10 by arendst