arendst/Tasmota #336: OTA Upgrade error (at a size of more than 50%)

AlexTransit, 2017-04-06:
i define ds18x20 library
"Sketch uses 482,096 bytes (50%) of program storage space. Maximum is 958,448 bytes."
ota upgrade error "File upload buffer miscompare"
prorgam via UART - normal.
problem only  big file
if "Sketch uses 480,296 bytes" - normal prorgam via OTA

davidelang, 2017-04-06:
This is a known problem with using OTA, and why there is work being done to 
shrink the codebase. One thing you can do is to build two images, one with what 
you want, and a second with a minimal build. Flash the minimal build from the 
full build and then flash the full build with the minimual build and you should 
be ok.

David Lang

k

smadds, 2017-04-06:
I wonder if it's worth considering making every OTA upgrade a 2-stage one? The first stage could be really REALLY minimal - just to clear the existing code and start loading the main version. This would give us much more space to work with and maybe not have to resort to turning options on and off in the source.

Even better would be code that could prune itself back to the minimum as part of the OTA process to leave space for the new upload. Not sure if that's even possible, but it would certainly be helpful if it could be done.

davidelang, 2017-04-06:
well, if you compile the image yourself, you may very well find that you can 
turn things off so that you don't need to do a 2-stage upgrade, so it doesn't 
make sense to do it for every upgrade unconditionally.

smadds, 2017-04-06:
Yes sure, but that's a big if David.

I thought the point of Tasmota was to have a standard firmware that could be configured post installation. We seem to be going back to custom compilations again. Having a half & half solution with some things configured pre and some post compilation seems to me the worst scenario for maintenance and long-term sustainability of the project.

If you are worried about the number of flashes of the prom, would it not be possible for the OTA to check whether there is enough space for a direct upgrade first then switch to a 2-stage if required?

davidelang, 2017-04-06:
Even with Tasmota there has always been a size contraint and different 
compilation options (look at TLS support for one big example of something not 
normally compiled in)

I created issue #217 to discuss options for dealing with the size issue.

Since then the webserver has been cleaned up a bit, but there are still size 
issues.

It's not just the number of flashes, it's the problem of uploading images to 
flash, and the naming of the images in the webserver and the software (and as 
always, the question of how minimal should the small image be)

arendst, 2017-04-06:
@davidelang there won't be much more clean-up possible; library updates are always bigger than before. Once the new Arduino-esp8266 is released things get really worse as it's almost 40k bigger already...

I don't like it either having all kinds of versions around but I suggest just to keep one version available via @smadds and let the users compile their own ones with optional features. 475k is just too small to have many features available,

davidelang, 2017-04-06:
Is it worth revisiting the ideas I raised in #217 about implementing a loadable 
module for all non-essential stuff? or should we forget it for now and just work 
on documenting the two-stage process for people?

  On Thu, 6 Apr 2017, Theo Arends wrote:

> @davidelang there won't be much more clean-up possible; library updates are always bigger than before. Once the new Arduino-esp8266 is released things get really worse as it's almost 40k bigger [...]

arendst, 2017-04-06:
Using the Arduino-esp8266 makes partitioning code almost impossible so I suggest to focus on the two stage approach.

stefanbode, 2017-04-06:
The esp8266 is awesome cheap and therefore limited in capacity. Even if we optimize we will hit the boundaries because then the expectation will be higher. Instead of crunching the 8266 more and more you should think about the next generation esp32. Same manufacturer, same concept, and much more power. Price is 10$ right now. I already use this one for the more complex operations.

davidelang, 2017-04-06:
the ESP-32 is a nice chip, but it suffers from one fatal flaw, it's not what's 
being shipped in devices (especially the sonoff devices) that are supported by 
this firmeware

In addition, there are still some missing libraries

On Thu, 6 Apr 2017, stefanbode wrote:

> The esp8266 is awesome cheap and therefore limited in capacity. Even if we optimize we will hit the boundaries because then the expectation will be higher. Instead of crunching the 8266 more and [...]

davidelang, 2017-04-06:
FYI #170 was opened to track support possibilities for the esp32

ionciubotaru, 2017-04-06:
@AlexTransit disable webserver and you will get lot of RAM and lot of free space on flash
You can send all commands via Mqtt

AlexTransit, 2017-04-07:
Thanks for the recommendation.
I ran into a problem.
And shared it with others.

closed 2017-04-23 by davidelang