arendst/Tasmota #337: Switch Mode not setting correctly

gid204, 2017-04-06:
I keep having this issue where my switch mode will not set correctly.
I am trying to achieve the use of a toggle button. So that I can hook up the sonoff to an electric gate and control it via mqtt commands.

Whenever I set the switch mode to anything it stays a mode 1.

04:07:41 CMND: cmd/sonoff-3/switchmode
04:07:41 MQTT: stat/sonoff-3/RESULT = {"SwitchMode1":0}
04:07:48 CMND: cmd/sonoff-3/switchmode2
04:07:49 MQTT: stat/sonoff-3/RESULT = {"SwitchMode2":0}
04:07:53 CMND: cmd/sonoff-3/switchmode
04:07:53 MQTT: stat/sonoff-3/RESULT = {"SwitchMode1":0}
04:07:59 CMND: cmd/sonoff-3/switchmode3
04:08:00 MQTT: stat/sonoff-3/RESULT = {"SwitchMode3":0}
04:08:05 CMND: cmd/sonoff-3/switchmode
04:08:05 MQTT: stat/sonoff-3/RESULT = {"SwitchMode1":0}
04:08:15 CMND: cmd/sonoff-3/switchmode4
04:08:16 MQTT: stat/sonoff-3/RESULT = {"SwitchMode4":0}
04:08:20 MQTT: stat/sonoff-3/RESULT = {"POWER":"OFF"}
04:08:20 MQTT: stat/sonoff-3/POWER = OFF
04:08:21 MQTT: stat/sonoff-3/RESULT = {"POWER":"ON"}
04:08:21 MQTT: stat/sonoff-3/POWER = ON
04:08:24 CMND: cmd/sonoff-3/switchmode
04:08:25 MQTT: stat/sonoff-3/RESULT = {"SwitchMode1":0}

I don't know if I am miss interpreting the commands but I just need to know how to get it set to toggle instead of 'follow'.

thanks in advance!

davidelang, 2017-04-06:
you are never setting the switchmode, you are only querying it

try

switchmode2 2

to set the mode for switch2 to 2 instead of 0

David Lang

gid204, 2017-04-06:
thank you so much haha I feel stupid but glad I understand and have it working now :D

closed 2017-04-23 by davidelang

anthonykeane, 2017-07-18:
Hi Guys
I am having the same issue with switchmode 

Program version 5.2.4
Build Date & Time 2017-07-03T09:51:15
Core/SDK version  2_3_0/1.5.3(aec24ac9)

switchmode2 2
11:18:26 CMND: switchmode2 2
11:18:26 MQTT: stat/daddy1/RESULT = {"Command":"Unknown"}

davidelang, 2017-07-18:
do you have two switches defined?

anthonykeane, 2017-07-19:
I have a the basic Sonoff (one relay, led, 5 pin header), I have since discovered that to use this command you need to set the GPIO14 (or other) to Switch1 only then will the command work

pprotschka, 2017-07-20:
thanks for the tip with setting any user defined pin to switch. this did indeed work. This is a bug in my opinion. this is not expected behaviour...

davidelang, 2017-07-20:
switchmode only works on switches, not on buttons, so if you don't have anything 
defined as a switch, the switchmode command will fail.