arendst/Tasmota #338: I can provide extension for I2C ADS1115. Intrested?

stefanbode, 2017-04-06:
I created a new file xsns_ads1115.ino linked on demand to support 16bit 4-channel A/D converter. Mr. Arend, interested in adding. Library used: Adafruit_ADS1X15-master

Code +1kb, Ram +968Byte.

ionciubotaru, 2017-04-06:
Test your code carefully and add a pull request. @arendst will decide or not to include it in the next release. Adding new features is good for community.

PitPaul, 2017-04-15:
Hey , can you send code to test it ?

stefanbode, 2017-04-15:
Take a look at the forks that where created. You should easily find mine with the additions. Let me know if you got problems. worked for me excellent. I do not transfer the voltage, just the pure digital output. Any postprocessing should be done in the application.

davidelang, 2017-04-23:
please provide a link to the fork in question

davidelang, 2017-04-23:
or ideally, create a pull request for the proposed changes.

stefanbode, 2017-04-23:
Hi David. You can reach me on skype. stefan.bode. The fork you will find here: https://github.com/stefanbode/Sonoff-Tasmota/tree/master/sonoff

davidelang, 2017-04-23:
taking a _very_ quick look at this, could you please create a couple branches with subsets of these changes

1. back out the user_config.h changes (I suggest using user_config_override.h locally so that you don't keep fighting this issue)

2. you have added several libraries. I believe that we would prefer to not have the libraries in our git tree and instead add them to the prereq list so that they are maintained and updated outside of tasmota. We only include libraries when we have a need to use a specific version and don't want to have them updated automatically.

3. it looks like you are adding two different sensors here. Having a separate branch that is specific to each sensor makes it easier to review and merge

Once you do this cleanup, please create a pull request for each of these sensors and in the pull request, document how much additional flash and ram each one adds

fdsailer, 2017-11-04:
Hello Everybody
I would be very interested into the support for the ADS1115. Is there a specific plan to support or has the good idee vanished ? If somebody can give me a insight ?

Thanks

Michael

closed 2017-11-07 by arendst