arendst/Tasmota #339: Multiple DHT sensors

HellGL, 2017-04-06:
Hello,
as follow up to issue #308 I created a patch to access up to 3 independent DHT sensors. I can't say how much additional ram or flash is used, but probably only a few bytes. Comments welcome.

[multi_dht2.patch.gz](https://github.com/arendst/Sonoff-Tasmota/files/902551/multi_dht2.patch.gz)

ionciubotaru, 2017-04-06:
Test your code carefully and add a pull request. @arendst will decide or not to include it in the next release

HellGL, 2017-04-07:
I am not sure if this correct, but at the moment I am sending all DHT sensor readings in one mqtt message:
`13:37:46 MQTT: tele/sonoff/SENSOR = {"Time":"2017-04-07T13:37:45", "DHT22_04":{"Temperature":19.4, "Humidity":42.7}, "DHT22_14":{"Temperature":19.8, "Humidity":43.5}}` 
Should I change the code to send a message for each sensor?

arendst, 2017-04-07:
No keep it on one line please

davidelang, 2017-04-07:
all sensors (of all types if there are multiple types) are sent as one sensor 
line.

davidelang, 2017-04-23:
If you can create a pull request, it's easier to discuss and comment on the proposed changes. If you are not able to do that (or don't want to), please let us know and we'll fall back to less efficient processes.

HellGL, 2017-05-02:
Now that I am back from vacation I added a pull request based on the latest (5.0.1) version.

arendst, 2017-05-03:
Implemented in next release

closed 2017-05-05 by arendst