arendst/Tasmota #340: Cannot send commands to sonoff basic if topic contains /

sebastianlorenzen, 2017-04-07:
I have configured my sonoff basic with the following topic:
sonoff/basic_01
When I send cmnd/snonoff/basic_01/POWER ON I get back an error "Unknown command"

Wen I rename the topic on the sonoff to only basic_01
The command cmnd/basic_01/POWER ON works as expected.

Is this by design?

I want to build structures of devices with he topic eg. snonoff/floor1/bed_room/light_01

davidelang, 2017-04-07:
inmqtt '/' has special meaning, you cannop use it in a topic

On Fri, 7 Apr 2017, sebastianlorenzen wrote:

> Date: Fri, 07 Apr 2017 01:37:21 -0700
> From: sebastianlorenzen <notifications@github.com>
> Reply-To: arendst/Sonoff-Tasmota
>     <reply+0021b24f1f590c3fd854f121ac20ec5635030865f6c9fc9892cf0000000114ff11c
>     192a169ce0d1f2e9f@reply.github.com>
> To: arendst/Sonoff-Tasmota <Sonoff-Tasmota@noreply.github.com>
> Cc: Subscribed <subscribed@noreply.github.com>
> Subject: [arendst/Sonoff-Tasmota] Cannot send commands to sonoff basic if
>     topic contains / (#340)
> 
> I have configured my sonoff basic with the following topic:
> sonoff/basic_01
> When I send cmnd/snonoff/basic_01/POWER ON I get back an error "Unknown command"
>
> Wen I rename the topic on the sonoff to only basic_01
> The command cmnd/basic_01/POWER ON works as expected.
>
> Is this by design?
>
> I want to build structures of devices with he topic eg. snonoff/floor1/bed_room/light_01
>
>

sebastianlorenzen, 2017-04-07:
Ah, good to know. Any best practice to structure topics with multiple devices? Maybe this should be added to the wiki.
Where do I do the structuring if not in the device?

davidelang, 2017-04-07:
topics are limited in sizr (33 characters IIRC) and cannot contain a /, #, + but 
it can contain other characters (I wouldn't use ' '), so you can use - etc

k

sebastianlorenzen, 2017-04-07:
maybe I got it wrong how to use topics.
Here are some best practices around topics I found:
http://www.hivemq.com/blog/mqtt-essentials-part-5-mqtt-topics-best-practices
https://tinkerman.cat/mqtt-topic-naming-convention

So my question: how to use Sonoff with Tasmota in such a setup?

Can this be implemented, that the topic allows slashes?

arendst, 2017-04-07:
Sonoff supports <cmnd>/<devicename>/<command> <action> where you can specify
- <cmnd> up to 10 characters long including a slash.
- <devicename> up to 32 characters long without slashes
- <command> recognized by sonoff and documented on wiki - command
- <action> to be performed by command.

That's my decision and it's final (for now).

sebastianlorenzen, 2017-04-07:
fine for me. Just did not know it is implemented like this.
Will use other elements to structure the devices.

closed 2017-04-07 by sebastianlorenzen