arendst/Tasmota #342: How to use the three switch mode?

1043717432, 2017-04-08:
Sonoff firmware has been programmed to write successfully, how to open the three switch mode?
May I ask how the e8266 firmware can be modified using three switch modes?

davidelang, 2017-04-08:
what do you mean by "the three switch mode"?

1043717432, 2017-04-08:
Modify the relay?

Satellietje, 2017-04-08:
Do you mean this?
3 additional relay at one sonoff basic.
on GPIO4 I have a temperature sensor, but it can also a watersensor or whatever.

[image]

[image]

1043717432, 2017-04-08:
@Satellietje 

Hello, powerful! Can you provide detailed contact maps? Temperature and humidity is how to connect ?
[image]
[image]
I am to modify the e8266 programming sonoff, how to set the parameters in the Module sonoff parameters, I am a three touch switch.

Satellietje, 2017-04-08:
This is GPIO4
[image]

[image]

[image]

[image]

AlexTransit, 2017-04-08:
[image]

1043717432, 2017-04-08:
@AlexTransit Hello, what is this?

1043717432, 2017-04-08:
@Satellietje Hello, thank you! I try to buy accessories!

AlexTransit, 2017-04-08:
@1043717432 Hi. I remove led and use free gpio12.  :)

Satellietje, 2017-04-08:
What's that for a switch in the picture

1043717432, 2017-04-08:
@AlexTransit  Hello, your picture is not clear, call gpio12 can do?

1043717432, 2017-04-08:
@Satellietje  Hello, use e8266 single fire switch, no zero line. I have written to sonoff, but there is a problem, I do not know how to set up to support the function of 2 or 3 relay switch. I want to know!

1043717432, 2017-04-08:
@Satellietje  Hello, may I ask you the first picture of the sonoff connection 3 relays, a detailed connection diagram?

davidelang, 2017-04-08:
to set it up to support up to 4 relays, just wire the relays to GPIO pins and 
configure them as relay2, relay3, or relay4 (it can be done via the webserver in 
the module configuration page, or via commands, see the wiki for the commands)

1043717432, 2017-04-08:
@Satellietje  Hello, I have 2 switch and 3 switch 2 switch now, I can choose the sonoff basic parameter setting module, should be supported, but the 3 do not know how to choose the switch panel module parameters, after all, sonoff 4CH is 8285.

1043717432, 2017-04-08:
@Satellietje Hello, thank you for sharing!

1043717432, 2017-04-08:
@Satellietje After I go home, I'll upload 3 pictures.

davidelang, 2017-04-08:
switches are inputs, relays are outputs. if you want the sonoff to turn 
something on and off you need to set it to be a relay, not a switch

1043717432, 2017-04-08:
@davidelang  Hello, I am not sonoff, just write the sonoff firmware e8266,

davidelang, 2017-04-08:
I don't understand what you are saying.

1043717432, 2017-04-08:
@davidelang @Satellietje @AlexTransit  
[image]

davidelang, 2017-04-09:
that tells me nothing.

davidelang, 2017-04-23:
closing this issue as there has been no response in two weeks. If you are still working on this, go ahead and re-open the issue.

closed 2017-04-23 by davidelang

1043717432, 2017-09-16:
@AlexTransit  hi，You remove the LED, GPIO12 re-access to other LED lights can be dimmed?
How to achieve