arendst/Tasmota #344: Slampher blinks at 1 Hz when board button pressed

NelisW, 2017-04-08:
recent version of Tasmota 2017-04-09. Built/downloaded using PlatformIO.
Old version of Slampher V0.2 PCB. 
R20 shorted out before flashing and left in place afterwards (I don't need 433 RF).

PCB powered by sufficiently capable external 3.3V supply, AC not connected. FTDI not connected.
After reboot all MQTT control works as expected (if the button on the board is not pressed).

When I press the button on the board, the Slampher goes into 1 Hz blinking cycle. Once in the blinking cycle, the board does not respond to MQTT commands any more.

Note: R20 still shorted out. Should I remove the short?

NelisW, 2017-04-08:
Forgot to tell: I did this experiment with the switch coupled into Home Assistant and then also by shutting down Home Assistant, so the Slampher is completely on its own.

The serial output during this cycle is as follows, perhaps it will help.
20:29:01 MQTT: stat/sonoff/RESULT = {"POWER":"ON"}
20:29:01 MQTT: stat/sonoff/POWER = ON
20:29:03 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
20:29:03 MQTT: stat/sonoff/POWER = OFF
20:29:04 MQTT: stat/sonoff/RESULT = {"POWER":"ON"}
20:29:04 MQTT: stat/sonoff/POWER = ON
20:29:05 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
20:29:05 MQTT: stat/sonoff/POWER = OFF

davidelang, 2017-04-23:
something is telling it to switch on and off, you can increase the debugging level and that may explain what is what. It could be a missing pull-up resister on your switch/button so that it's floating, but that wouldn't explain a steady 1 hz cycling.

closed 2018-05-08 by stale[bot]