arendst/Tasmota #348: Send PulseTime with On command

mrmin24, 2017-04-11:
Hi,

It would be nice to be able to override the PulseTime when sending a Power command, so that you can tell the relay to switch on for a specific time without overiding the pulsetime first with an additional command.  

e.g. /xxx/topic/ Power  ON_600   - irrigation goes on for 10mins

thanks for the great job

ionciubotaru, 2017-04-11:
Modify pulse time before power command. Send 2 commands instead of one.

mrmin24, 2017-04-15:
yes, but then you should check the pulsetime is set before sending second command.

ionciubotaru, 2017-04-16:
After the pulsetime setup you receive a confirmation from sonoff with the new pulsetime settings.
If you did't receive it send the first command again.

For multiple power commands with different pulsetime set the desired pulsetime before every command.

ascillato2, 2018-07-10:
Hi,

Seems that your question has been addressed. Closing this issue. Please reopen if needed. Thanks.

closed 2018-07-10 by ascillato2