arendst/Tasmota #349: Is there a POW Running/Stopped State based on threshold?

tysenmoore-xse, 2017-04-11:
I have an application (sump pump monitor) where I'd like to use the SonOff POW to track if something is "running" or not based on some current threshold.  So for example, if the current is greater than some threshold (maybe .5A) then assume "running" and if less than that it is assumed "stopped".  My application has a relatively short running time so I'd like to send the telemetry not based on the TelePeriod but rather on the state transitions--this way I can get more accurate run times and less MQTT updates.

I've looked at the commands and the user_conf.h but I don't see anything like this.  The CurrentHigh/CurrentLow seem promising but without code inspection I am not too sure of their actual use.

Is there support for either feature: "Running"/"Stopped" based on some threshold, or MQTT updates on transitions?  If not I plan to look at implementing this feature myself, maybe post a PR.  I figured I could save time if any portion of the use cases has already been implemented.  Thanks in advanced for any feedback.

davidelang, 2017-04-12:
you would do this logic in the home automation server that you have the sonoff 
sending it's mqtt messages to.
k

tysenmoore-xse, 2017-04-12:
Thanks for the feedback.  I have already implemented your suggestion for a washer and dryer within openHAB which but can have longer telePeriod with little impact.  However, for something with short "running" times in order to get reasonable data AND possibly not miss a "cycle" I'd have to reduce the telePeriod to something very short and hammer MQTT.  To reduce the amount of overhead and increase efficiency it seems like sending MQTT updates only on state transitions is a better approach--this also gives more accurate [start/stop] times--a low telePeriod will not compare to this implementation.  Or, am I not considering/missing something here?

jcullen86, 2017-04-12:
I too have been thinking about this and agree that having the sonoff send tele data triggered by a certain threshold is much better that having it send the data every 5ish second to get accurate data, especially when it may not be doing anything for most of the day.
Another option may be along the lines of "if watts is over 10w set tele period to 5sec otherwise set to 300sec".

davidelang, 2017-04-12:
one thing you are missing is that the current being drawn is only actually 
measured at the time the tele/ message is sent.

tysenmoore-xse, 2017-04-12:
I assumed that would be the case.  I have not looked at the specific code but it could easily track a previous value so it can detect transitions above or below the threshold and only on those transitions will it send tele messages.   The telePeriod at that point really becomes the polling period for checking the current--which could be much faster which only impacts internal Sonoff features (WebServer, etc).  However, I have not gone into the code yet.

arendst, 2017-04-14:
It takes at least 4 seconds for the HLW hardware to detect current, voltage and power so do not expect an immediate answer after power on/off your device.

tysenmoore-xse, 2017-04-14:
Thanks for or the details, that is good know.  One question, is that only at startup, or is that for every current/etc interrogation?

davidelang, 2017-04-23:
that is for every reading.

tysenmoore-xse, 2017-04-23:
Thanks so much for the information.  That could be a problem for my use case which at times runs less than 4 seconds so its "running cycle" could be missed.  Thanks again.

closed 2017-04-23 by tysenmoore-xse