arendst/Tasmota #351: Change Topic on minimal firmware

mrmin24, 2017-04-12:
It seems that if you change the MQTT topic while only running the **minimal** firmware, the topic is not saved and on restart the old topic is still used.

ionciubotaru, 2017-04-13:
It is not only the MQTT topic, last version 4.1.3 doesn't save parameters at all.
On booting I found the message:

`Config: Use default configuration
`

Previous versions works fine

arendst, 2017-04-14:
As designed.

Excerpt from user_config.h:
```
/*********************************************************************************************\
 * Compile a minimal version if upgrade memory gets tight ONLY TO BE USED FOR UPGRADE STEP 1!
 *   To be used as step 1 during upgrade.
 *   Step 2 is re-compile with option BE_MINIMAL commented out.
 *   !!! Needed for next release of Arduino/ESP8266 (+22k code, +2k mem) !!!
\*********************************************************************************************/
```
Not to be used as production environment but only for loading the bigger one!

ionciubotaru, 2017-04-14:
Please @arendst write few lines in the [upgrade section](https://github.com/arendst/Sonoff-Tasmota/wiki/Upgrade) of wiki. It is a little bit confusing how two steps OTA upgrade works.

arendst, 2017-04-14:
Just did. Hope it makes sense.

ionciubotaru, 2017-04-15:
@mrmin24 please close the issue.
@arendst please allow me to close issues

mrmin24, 2017-04-15:
Thanks anyway

closed 2017-04-15 by mrmin24