arendst/Tasmota #352: Relais module High Low Trigger

Satellietje, 2017-04-13:
I have expand some sonoff basic modules with relay modules.
[image]

I noticed that when the toggle on the sonoff is Off the relais module using 0.6 Watt Power. When the toggle is On the module use 0.05 Watt.
[image]

 I think it has to do with high and low level trigger.

I've looked and I saw the command SwitchMode, but this comand on te sonoff console displays:
MQTT: stat/sonoff/RESULT = {"Command":"Unknown"}

Does anyone know how I can turn the operation of the relay module so that when the toggle is Off the relay is at rest. I can just use the NC instead of NO to switch.

davidelang, 2017-04-13:
it's switchmode1 not just switchmode (for the first switch)

also note that switchmodeN is for switches and buttonmodeN is for the buttons. 
Buttons are the built-in buttons on the sonoffs, and switches were designed for 
added switches.

Satellietje, 2017-04-13:
What am I doing wrong?

[image]

davidelang, 2017-04-13:
what does your module configuration look like? (/md via the web interface, configuration -> module)

Satellietje, 2017-04-13:
Thanks for the tip!
There I found the solution :-)

closed 2017-04-14 by Satellietje