arendst/Tasmota #353: More than 4 relays

mrmin24, 2017-04-13:
Would it be possible to allow for up to 8 relays and pulse timers on devices such as the Wemos?

I did manage to get the device to think it has 8, but it does not switch the correct GPIOs

I added 

<code> else if (sysCfg.module == WEMOS) {
    Maxdevice = 8;
    Baudrate = 19200;
  } </code>

after the
<code> else if (sysCfg.module == SONOFF_SC) {
    Maxdevice = 0;
    Baudrate = 19200;
  } </code>

and added relays up to nr 8 in sonoff_template.h

Did not update MaxPulsetimers yet- might make a difference

davidelang, 2017-04-13:
There are quite a few places that you would need to modify in the current code 
to add additional relays. Including sonoff_template.h

I would love to see things rearranged so that instead of having to list every 
possible thing in enum upins_t { we could list all the types of things there, 
and then allow them to be selected multiple times (including relays, buttons, 
switches). but this would be a major rearrangement of code, probably close to a 
complete rewrite.

On Thu, 13 Apr 2017, mrmin24 wrote:

> Would it be possible to allow for up to 8 relays and pulse timers on devices such as the Wemos?
>
> I did manage to get the device to think it has 8, but it does not switch the correct GPIOs
>
> I added [...]

stefanbode, 2017-04-15:
I just received an 8 channel I2C to GPIO module. Will write an extension for it and for sure make the required changes in the code. Will have it working not later than end next week. It's my irrigation system that is now only missing the controller . Therefore a bit pressure to get it working.

My experience to get 8 relays working on the Wemos is negative. You have to use GPIO that are already predefined. Therefore I do not suggest to use 8 relays direct with the wemos

mrmin24, 2017-04-15:
Great news. I am going to plan for this then. Thanks

closed 2017-04-15 by mrmin24

stefanbode, 2017-04-20:
My fork is online and even if there are some limitation, it is a great step ahead. Leave me a note, if there are issues.
https://github.com/stefanbode/Sonoff-Tasmota/tree/stefan_patch1

mrmin24, 2017-04-20:
Thanks a lot thus far. 

I am not sure what I am doing wrong / should be doing

I connected the PCF8574 to the Wemos, loaded the new firmware, configured the device to be a Wemos and set up GPIO4 as SDA and GPIO5 as SCL. Also set up as Hue emulation

How do I now tell it that the PCF8574 has 8 relays connected?

This is my mqtt messages I get:

 {"Time":"2017-04-20T22:03:47", "Uptime":0, "Wifi":{"AP":1, "SSID":"Home", "RSSI":100, "APMac":"AC:9E:17:95:AD:10"}}

 {"Time":"2017-04-20T22:03:47", "AnalogInput0":5}

stefanbode, 2017-04-21:
Please can you post this in my Github and I will help or send me your skype ID or mail my is stefan.bode

mrmin24, 2017-04-21:
@stefanbode I cannot seem to post issues on your repo. I did see the troubleshoot update and will try that tonight. Thanks again.