arendst/Tasmota #354: Sonoff POW

ionciubotaru, 2017-04-13:
I set MaxPower = 500W and connect a 1kW heater for testing.
When the MaxPower is reached for MaxPowerHold seconds the relay is turned off, then after MaxPowerWindow seconds it is turned on again and the cycle repeats for MAX_POWER_RETRY times.

Everything works fine, except that this cycle cannot be stopped pressing the toggle button on web interface, nor send `cmnd/sonoff_pow/power 0`, nor pressing the button.

davidelang, 2017-04-13:
I think you would need to hit the button shortly after it turns on (probably 
within 1/2 maxpowerwindow to keep it from triggering), otherwise it's responding 
to the timers.

This is a good reason to have this logic done on your automation server, not in 
the device itself.

ionciubotaru, 2017-04-13:
It's not working.
I fired `cmnd/sonoff_pow/Power 0`  just after I received `stat/sonoff_pow/WARNING {"MaxPowerReached":"916"}` and the cycle continues. 
The only solution I found is `MAX_POWER_RETRY = 0`, in this case the cycle stops after first MaxPowerReached

davidelang, 2017-04-13:
try firing it off just after it resets (turns itself back on), not after it 
turns itself off.

On Thu, 13 Apr 2017, Ion wrote:

> It's not working.
> I fired `cmnd/sonoff_pow/Power 0`  just after I received `stat/sonoff_pow/WARNING {"MaxPowerReached":"916"}` and the cycle continues.
> The only solution I found is `MAX_POWER_RETRY = 0`, in this case the cycle stops after first MaxPowerReached

ionciubotaru, 2017-04-13:
I tried to send power off command after it automatically turns off and after it automatically turns on and none of these stop the cycle. Once protection cycle  is started there is no command to stop it

closed 2017-08-31 by ionciubotaru