arendst/Tasmota #355: Enable I2C configuration in web interface with Sonoff Pow

smmadge, 2017-04-13:
GPIO3 and GPIO4 seem to be unused with the Sonoff Pow. Is it possible to allow assigning them for use in I2C in the web interface? At the moment adding devices isn't possible with this specific Sonoff model. It would be very nice to add a BMP180 and SI7021 :-)

smadds, 2017-04-13:
It's been said before, but there are big potential safety issues with using sensors on the POW, @smmadge.

Unlike the other Sonoffs, this one has the logic circuitry tied to one leg of the incoming mains (should be neutral, but that's not always a certain). 

That said, I have mounted a BMP180 board within the Sonoff box - a couple of mm behind the hole in the side. I would only use this in places where it is out of reach of prying little fingers. I used the programming header for the sensor with some push-on connectors to allow reprogramming if everything else failed.

If you are sure about the safety aspects, the trick is to configure the unit as a Sonoff basic, configure the ports (and test), then reconfigure it to be a POW. Although you cannot see or change the sensor configuration, they still seem to work and show on the unit's homepage.

smmadge, 2017-04-14:
Thx! I'll make sure the sensors are placed in a secure spot!

smmadge, 2017-04-15:
Apparently this behaviour has changed in a more recent firmware version: using serial gpio1 as scl and gpio3 as sda works excellent in basic module configuration, but when switching to sonoff pow the configuration is lost. The console does not allow me to configure gpio at all.

davidelang, 2017-04-15:
you will need to manually edit the module configuration in sonoff_template.h

while it is possible to do this safely, it's hard, and anyone who can do it 
safely should have no trouble compiling their own build.

@arendst is not going to change the config in the build to encourage people to 
connect any sensors to the sonoff pow, it makes it an "attractive nuisance"

smmadge, 2017-04-15:
I can imagine that @arendst is not going to change the config. But it works now with the changes you mentioned! Thanks for your help!

closed 2017-04-15 by smmadge