arendst/Tasmota #356: Relay state at power up

mrmin24, 2017-04-16:
Hi,

I am not 100% sure I am doing everything correctly, but I have active low relays connected to a WEMOS. The relays are switched with relay1Inv and relay2Inv. 

At power up the relays activate for a split second until they are switched off by the firmware. Is there a way to prevent this from happening?

thanks

markbac17, 2017-04-16:
Hi,
I had almost the exact same scenario except the Wemos rebooted and eventually failed to turn on. I thought it was a power issue but I see that you have a similar experience.
I’ll play around a bit more but now I only activated 1 relay instead of 2.
Mark

DaNii78, 2017-04-17:
In my case this happens when the internet goes away. And also, when it comes back, the webserver doesn't work anymore.

I don't know why this is happening...

Thanks!

ionciubotaru, 2017-04-17:
I noticed this behavior and all Sonoffs:
- set the default relay to OFF on boot (works fine)
- if sonoff has un unexpected restart (not a software restart, nor hardware after a power loss), a software failure after multiple MQTT retries, the relay remains ON while sonoff reports it as OFF via MQTT or web interface. It is nice that after un unexpected restart your devices remains ON and I think the a digitalRead() will fix the relay status.

DaNii78, 2017-04-17:
In my case, it's configured to remember the last status before power off or restart itself.

But I don't know why sometimes, when the power is set to OFF, it turns ON during less than a second. 

Maybe could it be due to the state of the group topic?

stefanbode, 2017-04-20:
The problem in the INVERTED state. It works fine with the other type. The machine boot state is OFF. Therefore, if INVERTED it starts with a short ON time before going to OFF. This issue is also seen on my DIO I2C converter PCF8574. You can come around this problem by avoiding the INVERTED. This can be done by using one of the INVERTER IC like the ULN2803A. This solves the problem physically. There is no chance to fix this in software.

mrmin24, 2017-04-20:
OK, so after investigating the schematic of the WEMOS and testing some more I found that the GPIO 0,2 have pull ups on board and GPIO have a pull down. 

When connecting the 4 relays to GPIO 4,5,12,13 I do not see the problem of relays activating on power up. Remember inverted relays. So the problem goes away if you use the correct pins. Can probably use the TX/RX pins if serial debugging is off as well. 

The lights do flash on the relay board for a split second on power up but it is probably due to floating lines, but relays are not activated

Let me know if you agree and I will close the issue.

davidelang, 2017-04-26:
That sounds correct.

closed 2017-04-26 by davidelang

asantos15, 2018-02-27:
Hello @mrmin24 ,

Mine are connected over my wifi network and this turn on at boot is happening. For the one that is supposed to work as a garage door opener, this makes it unusable.
Would there be any solution for that at sight? I've been reading about it for weeks and I am running out of options.

Thanks in advance!

mrmin24, 2018-02-27:
@asantos15 I have been out of this for a little while, but I belive this was fixed in one of @stefanbode fork releases. I am not sure if it was ever fixed in the trunk.

stefanbode, 2018-02-27:
Yes, this is a side effect where I and arendst disagree how it should work. Depending on the wireing you get a wrong result with the original code. In sonoff.ino near line +2097 I made a small change to prevent an override of the status. if the code executes the digitalRead in your case you're in trouble.

```
    } else {
      //STB mod
      if ((pin[GPIO_KEY1 +button_index] < 99) && !blockgpio0 && button_index < MAX_KEYS) {
      //end
        button_present = 1;
        button = digitalRead(pin[GPIO_KEY1 +button_index]);
      }
    }
```

asantos15, 2018-02-27:
I did run the following on my pi yesterday and it seems to have solved the flickering problem:
mosquitto_pub -h localhost -p 1883 -r -t 'cmnd/sonoff/POWER' -n -u USERNAME-P PASSWORD

I will check it for a while to see if it doesn't get back to flickering...

Thanks guys!

nickidw, 2019-05-01:
I see there is a command PowerOnState that controls what the relay does: https://github.com/arendst/Sonoff-Tasmota/wiki/PowerOnState-Configuration