arendst/Tasmota #357: Wifi name issue

azazzel, 2017-04-16:
no caps for wifi ssid??? I cannot connect since the wifi ssid got capital "K"

logs: 
#define STA_SSID1              "Koanga"          // [Ssid1] Wifi SSID

00:00:07 Wifi: Connecting to AP1 koanga in mode 11N as sonoff-5810...
00:00:14 Wifi: Connect failed as AP cannot be reached

ionciubotaru, 2017-04-17:
Please check your AP settings.
I have un AP named Public (capital P) and Tasmota works fine.

davidelang, 2017-04-23:
I also have multiple caps in my ssid. closing this since it doesn't seem to be the real problem. please re-open if you are still having problems.

closed 2017-04-23 by davidelang

afulki, 2017-12-30:
#define STA_SSID1              "**Afulki**"          // [Ssid1] Wifi SSID

00:00:00 WIF: Connecting to AP1 _afulki_ in mode 11N as sonoff-5388...

It seems to be an issue for me too, get the error connect failed as AP cannot be reached.

arendst, 2017-12-31:
@afulki No problem what so ever:
```
ssid1 Alfuki
14:12:46 CMD: ssid1 Alfuki
14:12:46 RSL: Group 0, Index 1, Command SSID, Data Alfuki
14:12:46 MQT: stat/wemos9/RESULT = {"SSId1":"Alfuki"}
14:12:46 CFG: Saved to flash at F4, Count 44, Bytes 1648
14:12:47 APP: Restarting

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
þ
00:00:00 APP: Boot Count 26
00:00:00 Project sonoff Wemos9 Oled (Topic wemos9, Fallback DVES_4E493C, GroupTopic sonoffs) Version 5.10.1a
00:00:00 DBG: Display model 2
00:00:00 CFG: Saved to flash at FB, Count 45, Bytes 1648
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Patch issue 2186
00:00:00 WIF: Connecting to AP1 Alfuki in mode 11N as wemos9...
```
but your problem is with changing parameters in user_config.h instead of just using online commands as I did above...

afulki, 2018-01-02:
Yes, I figured it out thanks, the issue was to do with the data being stored in NVRAM, even though I was changing the config header, it wasn't being persisted.  Thanks for the help.