arendst/Tasmota #358: 16K spiffs

ionciubotaru, 2017-04-17:
Very good improvement.
Is it possible to upload the new configuration via OTA or we must upload first via serial ?
Few words about replacing boards.txt in the Wiki section will be helpful for newbies.

arendst, 2017-04-17:
Please wait! Last night (?!?) I decided to make spiff even as small as 8k but before that the current config has to be moved up by 8k.

My planning is this:
- next release will virtually move config up by 8k placing it at end of spiffs.
- a new repository will be released withanadmusted linker script and 8k spiffs
- as far asI could test only serial upload will allow for introducing a new linker script

Doing it this way will allow for using more code space while keeping the current configs.

Implementing will take some time as i'm away this week.

arendst, 2017-04-17:
Btw I also have to find a way how to tell Travis to use an updated linker script...

arendst, 2017-04-24:
Plans changed once more. I'll abandon SPIFFS area all together and move settings to EEPROM area. This removes the double copy of settings in use today but will 32k code space.

Next release (4.1.4) will implement the move of settings. Future versions will implement the linker change but I first have to find a way to tell Travis to use another linker and it would be nice to implement the linker without having to serial reflash the sonoffs... To be contnued.

arendst, 2017-04-24:
Fixed travis. Did some live testing between linkers and it seems there is an upgrade path from 64kSPIFFS to no SPIFFS without serial reflash. Continue testing...

arendst, 2017-04-25:
Just released version 5.0.0 with need for new linker script also available from the site. This can be implemented 'on the fly' by webpage upgrade as noted in the ReadMe.

In the end another 32k of code space made available for fixing and updates.

closed 2017-05-04 by arendst