arendst/Tasmota #359: Wemos control external LED

flanshaw, 2017-04-17:
Great firmware, been flashing my sonoffs all easter.

I've a wemos D1, reading analog0, DHT11 etc all  working great.. 

Ive read and re-read and can't figure out how to set D2 on wemoss to high/low, to control an LED?
[image]

stumpped...

cobra85ec, 2017-04-18:
i need help with this also

davidelang, 2017-04-18:
what gpio is D2 on?

On Mon, 17 Apr 2017, flanshaw wrote:

> Great firmware, been flashing my sonoffs all easter.
>
> I've a wemos D1, reading analog0, DHT11 etc all  working great..
>
> Ive read and re-read and can't figure out how to set D2 on wemoss to high/low, to control an LED?
> [image]
>
> stumpped...
>
>

decologne, 2017-04-18:
Pin translation table for Wemos D1 mini:
https://www.wemos.cc/product/d1-mini.html

davidelang, 2017-04-18:
ok, D2 is GPIO4 according to this link, so set gpio4 to either a led or relay 
output and then you can control it.

David Lang

On Tue, 18 Apr 2017, decologne wrote:

> Pin translation table for Wemos D1 mini:
> https://www.wemos.cc/product/d1-mini.html

flanshaw, 2017-04-18:
Perfect. Set GPIO as relay1. Send cmnd/wemos/power1 and works..

Thanks

closed 2017-04-18 by flanshaw

cobra85ec, 2017-04-18:
how about controlling 2 relays..

davidelang, 2017-04-18:
just define the gpios as relay2/relay3/etc

this firmware can be set to manage 4 relays, 4 LEDs, 4 switches, etc.

David Lang

On Tue, 18 Apr 2017, cobra85ec wrote:

> how about controlling 2 relays..
>
>