arendst/Tasmota #360: Alexa's crying about unreachable hub

ostpol, 2017-04-18:
Hi,

if I tell Alexa to turn my sonoff on/off, she does what i want but's complaining about an unreachable hub.

_(To be honest, I'm not on 4.1.3.)_

Any others with this problem? Is 4.1.3 the solution?

Thanks,
T.

citrusfizz, 2017-04-19:
I find i have better success when i set the wifi host name to something similar or exact as my alexa wemo friendly name.

ostpol, 2017-04-19:
Nice, I'll try this.

The sonoff has ~60% in WiFi signal strength. This should be enough for a fairly stable connection, but the webinterface is slow or even unreachable. No surprise, that Alexa has some kind of probem.

The sonoff runs on 4.1.2. I'll update to 4.1.3 and I also set the log level to "3 Debug". Maybe this will reveal some related information.

edmogeor, 2017-04-23:
I found the best way to get Alexa working with these sonoffs is to use a Raspberry Pi as a MQTT broker with Home Assistant installed also. This gives the added bonus of a nice webapp that you can access from your phone to turn the switches on and off. Bruh Automation on YouTube has an excellent guide to setting them up with MQTT. I had exactly the same issue as you but using the raspberry pi made it much more stable for some reason. It was pointed out to me that disabling mqtt in the code was also important. Good luck

goncalvesdiogo, 2017-05-16:
try to turn off the log mode. I believe that it tries to save something at the server, and it stucks when can't find it.

mohaggag, 2017-06-21:
how did you guys make it to work ,, i have more than one Sonoff device and all are flashed with the 5.2 firmware but when i ask alexa to discover my devices ,, the answer is  i couldn't find any smart home devices

closed 2018-05-08 by stale[bot]