arendst/Tasmota #362: WifiManager AP name: sometimes ITEAD instead of Sonoff

KonstantinMastak, 2017-04-18:
Sometimes (irregularly, maybe one of three or four tries) when I activate WifiManager AP mode my device brings access point with name "ITEAD-XXXX" instead of "Sonoff-xxxx". 

Did anybody face this except of me? How do you think, where can ESP keep this "ITEAD" constant if we change the flash?

ionciubotaru, 2017-04-19:
Your flash rewrite was incomplete, please write it again via serial

davidelang, 2017-04-23:
closing this issue. please re-open if you still have problems.

closed 2017-04-23 by davidelang

lichtteil, 2018-02-12:
@KonstantinMastak I ran now into the same issue with my Sonoff basic switch. Flashed successfully 3 times with some code alternations and the webserver works but sometimes the ITEAD AP appears and it won't connect to my wifi. Did you solve this somehow?

KonstantinMastak, 2018-02-12:
I noticed, that there are a lot of strange issues (maybe including the one you mentioned), if you try to use Sonoff right after you flashed it. You should remove power for several seconds and the re-power it again. Then it works much better. Soft reboot which follows flash doesn't help for some reason. Only hard reboot helps.

lichtteil, 2018-02-12:
@KonstantinMastak Thanks for the fast reply, I'll give it a try :)