arendst/Tasmota #364: Amazon Alexa and Sonoff 4CH

ulfa, 2017-04-19:
Hello, 

i use the software with a sonoff 4CH and i am able to switch the 4 switches via MQTT commands. 

Now i have Alexa at home, but i can only see the first switch (friendly name) in Alexa. The other three switches are not available in Alex. 

Does anybody has a solution, so i can see all four switches in Alexa?

davidelang, 2017-04-19:
change the emulation that you are using. one emulation only supports a single 
switch, the other acts as a hub with multiple switches behind it.

On Wed, 19 Apr 2017, Ulf wrote:

> i use the software with a sonoff 4CH and i am able to switch the 4 switches via MQTT commands.
>
> Now i have Alexa at home, but i can only see the first switch (friendly name) in Alexa. The other [...]

davidelang, 2017-04-26:
did this solve the problem for you?

ulfa, 2017-04-26:
sorry, but i am at vacation. I will test it, when i am home again.

ulfa, 2017-04-29:
Hi David,

now i am home again and tested the other emulation and this works fine.

Thank you for your help,
Ulf

closed 2017-05-05 by arendst