arendst/Tasmota #366: Button

raisude, 2017-04-20:
I have a sonoff basic. Gpio14 configured as Button2. Pushing the button noteingewiesen happens. Ehen configured as switch2 pushing the Button the Relais toggles. Releasing the button the Relais toggles again. 

What i have to do that configuration button2 funktions?

Regards Rainer

davidelang, 2017-04-23:
hitting button2 trys to switch relay2, if you don't have a relay2 to switch, you won't see anything happen.

what is it you are trying to have happen when you press the switch connected to gpio14?

raisude, 2017-04-23:
I have to momentanary Switches. Pressing ohne of them should do the same as the Button of the Sonoff. This means they should also toggle the Relay1.

davidelang, 2017-04-23:
then configure gpio14 as switch1 and change switchmode1 until you get the 
behavior you are looking for.

raisude, 2017-04-23:
I have done this. But doing this the Relais was switched off and on in unregular intervalls without doing anything.

davidelang, 2017-04-23:
Then you need to add a pull-up resistor on gpio14

raisude, 2017-04-23:
Which Kind of Resistor should this be?

davidelang, 2017-04-23:
a 4-10k resister is common, really just about anything will do. connect it 
between gpio14 and 3.3v

raisude, 2017-04-23:
Ok i will Text this what about the Second Switch? Momentary connected to on of the serial ports?

Also ohne resistor between the serial port and 3.3v. Of connecting the Second Switch parallel to the First?

davidelang, 2017-04-23:
what do you want the second switch to do?

raisude, 2017-04-23:
The same. Toggling Relais1.

davidelang, 2017-04-23:
then you need to wire it in parallel with the first switch, not to a seperate 
GPIO. This firmware doesn't allow multiple GPIOs to have the same functionality.
k

davidelang, 2017-04-26:
did this solve your problem?

raisude, 2017-04-26:
Sorry for not giving feedback. The sonoff is in italy and during the week i'm in Germany. In italy i Bad no resistor so i can test it at the Weekend. 

So far thanks for your support. I will report on saturday.

raisude, 2017-05-01:
Today I have tested the resistor between 3.3V and GPIO14. Seems that this solved the problem of the ghost switching ... .

Thanks for your support.

Regards Rainer

davidelang, 2017-05-01:
please close the issue if it's resolved.

closed 2017-05-01 by raisude