arendst/Tasmota #367: Request, esp8285 successful firmware, modify the GPIO position

1043717432, 2017-04-20:
Request, esp8285 successful firmware, know the location of the switch pins are gpio0, gpio4, gpio9, gpio10, how to modify the firmware so that you can use the 2 switches, respectively, the 3 switches, the use of the 4 switches?
Also want to access to the MQTT to send the control switch, but also access to temperature and humidity sensors to the gpio14, how to set the request firmware?
[image]

1043717432, 2017-04-20:
My chip is ESP8285, after the success of Sonoff-Tasmota, I found the touch switch pin gpio0 and gpio4, and the other 1 switches are 3 touch switch, pins are gpio0, gpio4, gpio9, I want to ask how to modify the firmware, so can the normal use of the touch switch!

davidelang, 2017-04-21:
This is all done in the module configuration

select the WeMos and you can then set any gpio to any function.

1043717432, 2017-04-21:
@davidelang  Thanks！！ I'll test it right away! Thank

1043717432, 2017-04-21:
@davidelang  Cannot start after modification!

davidelang, 2017-04-21:
see what shows up in the logs (serial console if nothing else)

you can always reset your config and try again. You may have attempted to use a 
GPIO that conflicts with something else on your board.
k

1043717432, 2017-04-21:
@davidelang This is the IP address of the 86 box switch on the network, which means that it may not start successfully! Unable to learn log content!

davidelang, 2017-04-21:
you can see the logs via the serial port (with the appropriate settings)

David Lang
k

1043717432, 2017-04-23:
@davidelang Query less, I use the Arduino ide firmware.

davidelang, 2017-04-23:
using the arduino ide firmware doesn't mean that you can't see the serial 
port. AsI understand it, the IDE has a serial monitor specifially for that (I 
use other software)

davidelang, 2017-04-26:
did you get this working?

1043717432, 2017-04-28:
@davidelang 
Please close the problem

closed 2017-04-28 by davidelang