arendst/Tasmota #368: MQTT & Domoticz

jeylites, 2017-04-20:
I'm new to Sonoff MQTT and Domoticz.  I'm unable to get them to talk. Is there a tool that I could use to sniff Receiving and Sending signals? 

Domoticz Logs 

```
2017-04-20 12:44:11.579 Domoticz V3.7369 (c)2012-2017 GizMoCuz
2017-04-20 12:44:11.579 Build Hash: e9e981d, Date: 2017-04-19 05:39:52
2017-04-20 12:44:11.580 System: Raspberry Pi
2017-04-20 12:44:11.580 Startup Path: /home/pi/domoticz/
2017-04-20 12:44:12.558 PluginSystem: Started, Python version '3.4.2'.
2017-04-20 12:44:12.574 Active notification Subsystems: gcm, http (2/12)
2017-04-20 12:44:12.610 WebServer(HTTP) started on address: :: with port 8080
2017-04-20 12:44:12.627 WebServer(SSL) started on address: :: with port 443
2017-04-20 12:44:12.628 Proxymanager started.
2017-04-20 12:44:12.631 Starting shared server on: :::6144
2017-04-20 12:44:12.631 TCPServer: shared server started...
2017-04-20 12:44:12.631 RxQueue: queue worker started...
2017-04-20 12:44:14.632 EventSystem: reset all events...
2017-04-20 12:44:14.634 EventSystem: reset all device statuses...
2017-04-20 12:44:14.635 EventSystem: Started
2017-04-20 12:44:14.665 PluginSystem: Entering work loop.
2017-04-20 12:44:15.632 MQTT: Connecting to 192.168.1.69:1883
2017-04-20 12:44:16.022 MQTT: connected to: 192.168.1.69:1883
2017-04-20 12:44:16.122 MQTT: Subscribed
2017-04-20 12:44:16.253 Incoming connection from: 192.168.1.78
2017-04-20 12:45:39.579 User: Admin initiated a switch command (1/sonoff-IDX1/Off)
2017-04-20 12:45:39.579 (sonoff-IDX1) Light/Switch (sonoff-IDX1)
2017-04-20 12:45:41.679 User: Admin initiated a switch command (1/sonoff-IDX1/On)
2017-04-20 12:45:41.679 (sonoff-IDX1) Light/Switch (sonoff-IDX1)
2017-04-20 12:47:01.880 User: Admin initiated a switch command (1/sonoff-IDX1/Off)
2017-04-20 12:47:01.881 (sonoff-IDX1) Light/Switch (sonoff-IDX1)
2017-04-20 12:47:56.944 EventSystem: reset all device statuses...
2017-04-20 12:48:06.515 User: Admin initiated a switch command (2/Sonoff/On)
2017-04-20 12:48:06.516 (sonoff-IDX1) Light/Switch (Sonoff)
2017-04-20 12:48:08.212 User: Admin initiated a switch command (2/Sonoff/Off)

```

davidelang, 2017-04-21:
tcpdump/wireshark can let you sniff your network, but you can see exactly what's 
happening on the tasmota device by setting the logging level to more debugging

k

jeylites, 2017-04-23:
any idea what the Tasmota logs mean. I got Domoticz working with MQTT but there are a few things I'm not sure on the Tasmota app. Do I have to make changes to the MQTT settings below? What about User & Password ?

MQTT Host	domus1
MQTT Port	1883
MQTT Client &
 Fallback Topic	00014052
MQTT User	mqtt
MQTT Topic	Sonoff Lite
MQTT Group Topic	sonoffs
Emulation	Belkin WeMo

**TASMOTA LOGS**
00:00:01 Wifi: Attempting connection...
00:00:02 Wifi: Checking connection...
00:00:02 Wifi: Attempting connection...
00:00:03 Wifi: Checking connection...
00:00:03 Wifi: Connected
00:00:03 mDNS: Initialized
00:00:03 HTTP: Webserver active on Sonoff Lite-7817.local with IP address 192.168.1.10
00:00:03 UPnP: Multicast (re)joined
00:00:03 UPnP: Multicast disabled
00:00:03 RTC: (UTC) Sun Apr 23 20:23:16 2017
00:00:03 RTC: (DST) Sun Mar 26 02:00:00 2017
00:00:03 RTC: (STD) Sun Oct 29 03:00:00 2017
21:23:18 UPnP: Multicast (re)joined
21:23:19 UPnP: Multicast disabled
21:23:19 MQTT: Attempting connection...
21:23:20 mDNS: Query done with 0 mqtt services found
21:23:26 MQTT: Connect FAILED to domus1:1883, rc -2. Retry in 10 seconds
21:23:26 HTTP: Handle root
21:23:27 UPnP: Multicast (re)joined
21:23:32 HTTP: Handle console
21:23:37 APP: Multipress 1
21:23:37 UPnP: Multicast disabled
21:23:37 MQTT: Attempting connection...
21:23:38 mDNS: Query done with 0 mqtt services found
21:23:45 MQTT: Connect FAILED to domus1:1883, rc -2. Retry in 10 seconds
21:23:45 RSLT: domoticz/in = {"idx":2,"nvalue":0,"svalue":""}
21:23:45 RSLT: stat/Sonoff Lite/RESULT = {"POWER":"OFF"}
21:23:45 RSLT: stat/Sonoff Lite/POWER = OFF
21:23:46 Config: Saved configuration (1364 bytes) to flash at F8 and count 115
21:23:46 UPnP: Multicast (re)joined
21:23:52 Wifi: Checking connection...
21:23:52 Wifi: Connected
21:23:56 UPnP: Multicast disabled
21:23:56 MQTT: Attempting connection...
21:23:57 mDNS: Query done with 0 mqtt services found
21:24:03 MQTT: Connect FAILED to domus1:1883, rc -2. Retry in 10 seconds
21:24:04 UPnP: Multicast (re)joined
21:24:14 UPnP: Multicast disabled
21:24:14 MQTT: Attempting connection...
21:24:15 mDNS: Query done with 0 mqtt services found
21:24:21 MQTT: Connect FAILED to domus1:1883, rc -2. Retry in 10 seconds
21:24:22 UPnP: Multicast (re)joined
21:24:26 Wifi: Checking connection...
21:24:26 Wifi: Connected

**CONFIGARATION** 
**S20 Socket Module**

Sonoff Lite

Program version	4.1.3
Build Date & Time	2017-04-24T04:16:35
Core/SDK version	2_3_0/1.5.3(aec24ac9)
Uptime	0 Hours
Flash write count	115
Boot count	28
Reset reason	Software/System restart
Friendly name 1	Sonoff Lite
 
AP1 SSId (RSSI)	SMILE (84%)
Hostname	Sonoff Lite-7817
IP address	192.168.1.10
Gateway	192.168.1.254
Subnet mask	255.255.255.0
DNS server	192.168.1.254
MAC address	5C:CF:7F:9F:3E:89
 
MQTT Host	domus1
MQTT Port	1883
MQTT Client &
 Fallback Topic	00014052
MQTT User	mqtt
MQTT Topic	Sonoff Lite
MQTT Group Topic	sonoffs
Emulation	Belkin WeMo
mDNS Discovery	Enabled
mDNS Advertise	Webserver
 
ESP Chip id	10436233
Flash Chip id	1327328
Flash size	1024kB
Program flash size	1024kB
Program size	465kB
Free program space	472kB
Free memory	27kB

davidelang, 2017-04-23:
so per the logs, it is failing to connect to your mqtt host (domus1), are you 
sure you have a mqtt server running there?

you only need to set user and password if you configured your mqtt server to use 
them

jeylites, 2017-04-23:
Here are my Domotiza logs and MQTT settings. The IDX 2 is the same on both ends...like wise on user and password.

[image]
[image]

terba, 2017-04-24:
You have to set "MQTT Host" on the tasmota device to the hostname or IP of your MQTT server. Try to connect to your MQTT server from an MQTT client (like mqtt-spy) from an other computer on the LAN.

CommodoreWhite, 2017-05-02:
Domus1 is tasmota's default name for the Mqtt server. I use domiticz too and host my Mqtt on the same Pi. Use tasmota Mqtt config web page and set the Mqtt host address there.

For maximum enjoyment, build your Pi using Peter Scargills script and install domiticz on it afterwards.

jeylites, 2017-05-03:
@terba @CommodoreWhite 

Thanks, guys for your advice, I set the "MQTT Sever" on Tasmota the same IP address as domotociz MQTT broker and it all worked.

jeylites, 2017-05-03:
@CommodoreWhite do you have any link/documentation on "Peter Scargills script" . Thanks in advance.

closed 2017-05-14 by arendst