arendst/Tasmota #369: Parameters changed in the web console are not persistent

garythebat, 2017-04-20:
Hi, I hope someone can help. I have tried reading, with no joy. 

In order to save some time I decided to preconfigure all my ssid/passwords/mqtt settings IP address etc in the config file.  I changed the value of the switch, built and uploaded with platform. 

Worked like charm. --except now no figures which I enter into the web console persist. 

Eg I want to set the domoticzidx1 to 3 but it just keeps defaulting back to one. Does anyone know what I have done here.  How do I get the figures entered into the web interface to persist.  I have already tried to fiddle with save data states - even those go straight to the defaults after restarting

davidelang, 2017-04-21:
see the troubleshooting page on the wiki, it explains this in detail
k

garythebat, 2017-04-21:
Thanks I've read it, but I cant see where it explains it, I did change the CFG_HOLDER value, and I have no problem with it saving the values from the config file, 

My problem is that when it is running and I log into the web-consol to change parameters they dont persist   

eg a domitczidx1 3 command returns a notice that the domitczidx1 state is 3 ,,   then it reboots and if you check the domoticzidx1 state it has reverted back to zero,

Also if you change the parameters in the little blocks in any of the config pages and save, it doesnt keep the settings after reboot

davidelang, 2017-04-21:
did you use be_minimal in your config? that disables saving any config changes
kk

garythebat, 2017-04-21:
I don't think so, I didn't modify that parameter in the config file. I will check that when I get home, thank you!

garythebat, 2017-04-21:
I am a complete NOOB at all things flashing and home automation, so this is very trial and error for me, 

Checked be_minimal wasn't enabled. No matter what I tried, if I changed the CFG_HOLDER and loaded with my parameters: any changes I then made in the web console would not persist. Tried it in 5 Sonoff's. before I gave up on that angle.

This wouldn't be a problem if I could have figured out how to set domitcz idx values in user config file, but I couldn't so I needed to be able to set them from WEB console

I eventually got around it by changing CFG_HOLDER then flashing with my parameters, then changing CFG_HOLDER back to original and flashing again -  then I could make changes in the web console and they would persist.

So I now have 8 sonoffs all working perfectly with Tasmota on Domoticz

davidelang, 2017-04-23:
since it sounds like you have it working, I'm going to close this issue. please re-open it if you are still having issues.

closed 2017-04-23 by davidelang