arendst/Tasmota #370: Request Support for Arilux RGB Strips

alexschwantes, 2017-04-21:
Saw Tasmota has support for the H801 LED strip. These strips are quite similar but cheaper and smaller esp8266 modules. http://www.banggood.com/search/arilux-rgb.html
[image]

[image]
There are a mix of RGB/RGBW/RGBWW units. Some have IR or RF as well. They all use a combination of gpios 5,4,12,13,14. Unfortunately the GPIO's are mapped to different functions between the different models.

As it is very similar to the H801 unit, I got it to work with only a slight mod of the sonoff_template.h for H801. With the following I can control it via PWM commands. Its not great as you have to send a different command for Red, Green and Blue values. 
```
  { "H801",            // Quick hack to support Arilux AL-LC03 (ESP8266)
     0,                // 
     GPIO_RXD,         // GPIO01 TX
     0,                // 
     GPIO_TXD,         // GPIO03 RX
     GPIO_PWM4,        // GPIO04 White
     GPIO_PWM2,        // GPIO05 Green
     0, 0, 0, 0, 0, 0, // Flash connection
     GPIO_PWM3,        // GPIO12 Blue
     GPIO_PWM5,        // GPIO13 White
     GPIO_PWM1,        // GPIO14 Red
     0, 0, 0
  },
```

Its not as nice as using https://github.com/mertenats/Arilux_AL-LC0X firmware that supports controlling the units via mqtt/IR/RF as it allows fading and setting colors with single commands. 

Sonoff LED Strip has RGB output as well on GPIO's 4/5/15 (See teardown https://www.orbit.me.uk/index.php/articles-all/14-technology/218-itead-sonoff-led-dimmer-photos). It would be nicer to control the Arilux in a similar way where you can fade between colors and set basic colors with one command. 

It was just a thought as the H801 was supported in a basic manner.

davidelang, 2017-04-23:
can you create a PR that includes the modifications to the template for each of the models?

alexschwantes, 2017-04-24:
Yeah I can do, but I'm not too pleased with the control of the LED. From my understanding you need to do 3 MQTT posts to update the RGB PWM values. Is that correct?

Is there anyway to chain updates of PWM commands in one MQTT packet?
But ideally it would be nicer to utilise the code for the sonoff led which has better functionality to set colours and fade between colours.

davidelang, 2017-04-26:
I would suggest looking at abstracting the color functionality out of the sonoff LED specifics and make it a more generic function that can be used for either 2 or 3 colors depending on what's defined.

BenNeumeister, 2017-11-03:
Hi @arendst ,

Just flashed a couple of my Arilux LC01 with 5.8.0q and not having much luck getting them to interfaces or even respond to the basic queries.

So the units don't recognise the 'LED' commands;
23:40:09 CMD: COLOR
23:40:09 MQT: stat/islandBench/RESULT = {"Command":"Unknown"}
23:40:15 CMD: Dimmer
23:40:15 MQT: stat/islandBench/RESULT = {"Command":"Unknown"}
23:40:20 CMD: Fade
23:40:20 MQT: stat/islandBench/RESULT = {"Command":"Unknown"}

same going via MQTT;
23:41:02 CMD: stat/islandBench/COLOR
23:41:02 MQT: stat/islandBench/RESULT = {"Command":"Unknown"}
23:41:16 CMD: stat/islandBench/DIMMER
23:41:16 MQT: stat/islandBench/RESULT = {"Command":"Unknown"}
23:41:21 CMD: cmnd/islandBench/POWER
23:41:21 MQT: stat/islandBench/RESULT = {"POWER":"ON"}
23:41:21 MQT: stat/islandBench/POWER = ON

Any ideas?
Should I try 5.9.0?

arendst, 2017-11-03:
Try command ``setoption15 1`` to enable color and dimmer

davidelang, 2017-11-03:
check your module configuration to make sure that it is configured to support 
these strips.

lgnd33, 2017-11-03:
These are supported under "MagicHome" name, but it's not known if @BenNeumeister configured his unit as such.
Also, original post (from April (!)) suggest there might be a different pinout for different types of devices. All units I own have same pinout...

arendst, 2017-11-03:
Starting with version 5.8.0p there is now also a specific Arilux LC01 module available. Just enable ``Setoption15 1`` and all should be fine.

BenNeumeister, 2017-11-05:
Hi @arendst ,
Yep that did the trick.
Feel like a tool for not looking over those options.

Appreciate the response!

closed 2017-11-05 by arendst